zephyrproject-rtos/zephyr #24277: tests/kernel/critical times out on ARC

andrewboie, 2020-04-10:
I have observed this on nsim_sem and nsim_em_mpu_stack_guard.

```
*** Booting Zephyr OS build zephyr-v2.2.0-1379-ga5cd799523a1  ***
Running test suite kernel_critical
===================================================================
starting test - test_critical
Start reg1 at 0
End reg1 at 5000
Start alt1 at 5000
End alt1 at 10000
Enable timeslicing at 10000
Start reg2 at 10000
End reg2 at 15000
Start alt2 at 15000
Time went backwards: 0 < 15750
Time went backwards: 0 < 3070
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
Time went backwards: 0 < 3370
Time went backwards: 0 < 3350
```

andrewboie, 2020-04-27:
This test code got moved to tests/kernel/workq/critical.
I can only get it to reproduce on nsim_sem now.

carlescufi, 2020-05-21:
@vonhust will you have time to take a look?

vonhust, 2020-05-25:
Fixed by #25403

closed 2020-05-25 by vonhust