zephyrproject-rtos/zephyr #24278: Function of "ull_conn_done" in "ull_conn.c"

BOZHENG001, 2020-04-10:
Hi All

I am right now developing BLE application based on `central_hr`and `peripheral_hr`on nRF52840 DK with zephyr 2.0.

I am now looking through the link-layer code from zephyr. A function called `ull_conn_done` in file `ull_conn.c` is a little confusing for me. I can see some basic comments in it. I suppose it should be about building connection for each connection event/interval. Also, there should be anchor point synchronization check for every connection event/interval. But I am not sure if I understand them right? Please correct me if anything wrong here.

Besides, I see a parameter inside the function `ull_conn_done` called `conn->supervision_expire`. Is it used to monitor the time offset for anchor point between the real one and the calculated one? Or to check if the anchor point is outside the `windowWidening` (a concept from BLE spec)? Or maybe some other events?

cvinayak, 2020-04-12:
Purpose of `ull_conn_done` 
- Check for MIC failures in connection event
- Handle connection termination procedure
- Manage anchor point drift in slave connection
- Handle connection failed to be established using `connect_expire` count down
- Handle supervision timeout using `supervision_expire` count down
- Handle control procedure timeout using `procedure_expire` count down
- Handle authenticated payload timeout using `apto_expire` and `appto_expire` count down
- Process connection RSSI sampled, store and generate events as necessary
- Manage slave latency changes due to data packets and control procedures
 
> I suppose it should be about building connection for each connection event/interval.

No, its is not able building connection event, but handling a completed connection event (by vendor specific Lower Link Layer, LLL)

> there should be anchor point synchronization check for every connection event/interval.

Anchor point synchronization is handled by vendor-specific LLL, and communicated in the `done->extra.trx_cnt` count.

> I see a parameter inside the function ull_conn_done called conn->supervision_expire. Is it used to monitor the time offset for anchor point between the real one and the calculated one? Or to check [...]

`supervision_expire` count down the consecutive intervals not receiving a packet or has CRC errors, it is used to perform disconnection  with reason supervision timeout.
Time offset calculation is vendor-specific, is supplied in `done->extra.slave.start_to_address_actual_us` by the LLL.

> I am now looking through the link-layer code from zephyr. 

If you plan to make changes to the design of the link layer for your application, you will void the Bluetooth qualified design listing for the controller and you will need to handle the qualification testing and process.

BOZHENG001, 2020-04-12:
Thanks for your answer and suggestion! 

One more question is about `supervision_expire`. As you mentioned, `supervision_expire` is used to count the consecutive intervals not receiving a packet or has CRC errors. If we only consider consecutive intervals not receiving a packet, can I also understand it like master and slave fail to synchronize the anchor point? 

Because if they fail to synchronize the anchor point at the beginning of connection event No.1, they will not able to arrange time for sending and receiving packets, and have to try to synchronize again in connection event No.2, which also means there will be no packets in connection event No.1. Then of course `supervision_expire` will count 1 for the situation above. Or BLE spec has some other rules for it?

cvinayak, 2020-04-14:
`supervision_expire` is implemented as a count down.

> If we only consider consecutive intervals not receiving a packet, can I also understand it like master and slave fail to synchronize the anchor point?

No. Packets with CRC errors are valid synchronization of anchor point. The controller will continue with the connection event unless two consecutive CRC errors.

Please refer to Bluetooth Specification for details.

BOZHENG001, 2020-04-14:
Thanks for your answer!

closed 2020-04-14 by BOZHENG001