zephyrproject-rtos/zephyr #24282: echo_client sample return: Cannot connect to TCP remote (IPv6): 110

overheat, 2020-04-11:
**Describe the bug**
I have tried echo_server sample, it works. And I leave it as the "server“，but got error after I start echo_client sample.

**To Reproduce**
Steps to reproduce the behavior:
Terminal #1
1. west build -b native_posix samples/net/sockets/echo_server
2. mv build build_server
3. ./build_server/zephyr/zephyr.exe
Terminal #2
1. west build -b native_posix samples/net/sockets/echo_client
2. mv build build_client
3. ./build_client/zephyr/zephyr.exe

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK, ...)
 - Commit SHA 6540f81f7ca2859d50a9c7d9b216ee5da04bd291

overheat, 2020-04-12:
I tried net-tools echo-server instead of  echo_server sample, but it is the same issue.
FYI.

jukkar, 2020-04-12:
In your setup zephyr will get following IPv4 addresses 192.0.2.1 (server), 192.0.2.2 (client) and your Linux host will get 192.0.2.2. So there is IPv4 address conflict. Simplest solution is to set zeth in your Linux host to have some other address like 192.0.2.3. Just go to net-tools directory, copy zeth.conf to zeth3.conf, and then edit zeth3.conf and set the addresses correctly. Then do `./net-setup.sh -c zeth3.conf`

overheat, 2020-04-13:
@jukkar Thanks. It make sense.
But I have tried posix<->posix configuration, which is server IP 192.0.2.1 and client IP 192.0.2.2, it return the same error number.
BTW, in linux<->posix case, [zephyr docs](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_client/README.html#running-echo-server-in-linux-host) told us 
```
$ sudo ./echo-server -i zeth
```
after that, we can see the echo-server is binded to linux IP, which is 192.0.2.2 by default. Looks I only need to modify my posix echo-client code as the comment in the source code. The same issue happens, even we do so.

Please help us to check it.

jukkar, 2020-04-15:
I managed to get echo-server<->echo_client working using two native_posix processes. The procedure is a bit complicated but basic steps were already described in
`samples/net/eth_native_posix/README.rst` file.

Create `net-tools/zeth-issue-24282-zeth1.conf` file
```
# Configuration file for setting IP addresses for a network interface.
INTERFACE="$1"
HWADDR="00:00:5e:00:53:11"
IPV6_ADDR_1="2001:db8:100::2"
IPV6_ROUTE_1="2001:db8:100::/64"
IPV4_ADDR_1="198.51.100.2/24"
IPV4_ROUTE_1="198.51.100.0/24"
ip link set dev $INTERFACE up
ip link set dev $INTERFACE address $HWADDR
ip -6 address add $IPV6_ADDR_1 dev $INTERFACE nodad
ip -6 route add $IPV6_ROUTE_1 dev $INTERFACE
ip address add $IPV4_ADDR_1 dev $INTERFACE
ip route add $IPV4_ROUTE_1 dev $INTERFACE > /dev/null 2>&1
```

Create `net-tools/zeth-issue-24282-zeth2.conf` file
```
# Configuration file for setting IP addresses for a network interface.
INTERFACE="$1"
HWADDR="00:00:5e:00:53:22"
IPV6_ADDR_1="2001:db8:200::2"
IPV6_ROUTE_1="2001:db8:200::/64"
IPV4_ADDR_1="203.0.113.2/24"
IPV4_ROUTE_1="203.0.113.0/24"
ip link set dev $INTERFACE up
ip link set dev $INTERFACE address $HWADDR
ip -6 address add $IPV6_ADDR_1 dev $INTERFACE nodad
ip -6 route add $IPV6_ROUTE_1 dev $INTERFACE
ip address add $IPV4_ADDR_1 dev $INTERFACE
ip route add $IPV4_ROUTE_1 dev $INTERFACE > /dev/null 2>&1
```

Terminal 1:
-----------

`./net-setup.sh -c zeth-issue-24282-zeth1.conf -i zeth.1`

Terminal 2:
-----------

`./net-setup.sh -c zeth-issue-24282-zeth2.conf -i zeth.2`

Terminal 3:
-----------
```
sudo brctl addbr zeth-br
sudo brctl addif zeth-br zeth.1
sudo brctl addif zeth-br zeth.2
sudo ifconfig zeth-br up
```

Terminal 4:
-----------
```
west build -d build/server -b native_posix samples/net/sockets/echo_server -- -DCONFIG_NET_CONFIG_MY_IPV4_ADDR=\"198.51.100.1\" -DCONFIG_NET_CONFIG_PEER_IPV4_ADDR=\"203.0.113.1\" -DCONFIG_NET_CONFIG_MY_IPV6_ADDR=\"2001:db8:100::1\" -DCONFIG_NET_CONFIG_PEER_IPV6_ADDR=\"2001:db8:200::1\" -DCONFIG_NET_CONFIG_MY_IPV4_GW=\"203.0.113.1\" -DCONFIG_ETH_NATIVE_POSIX_DRV_NAME=\"zeth.1\" -DCONFIG_ETH_NATIVE_POSIX_MAC_ADDR=\"00:00:5e:00:53:01\" -DCONFIG_ETH_NATIVE_POSIX_RANDOM_MAC=n
```

Terminal 5:
-----------
```
west build -d build/client -b native_posix samples/net/sockets/echo_client -- -DCONFIG_NET_CONFIG_MY_IPV4_ADDR=\"203.0.113.1\" -DCONFIG_NET_CONFIG_PEER_IPV4_ADDR=\"198.51.100.1\" -DCONFIG_NET_CONFIG_MY_IPV6_ADDR=\"2001:db8:200::1\" -DCONFIG_NET_CONFIG_PEER_IPV6_ADDR=\"2001:db8:100::1\" -DCONFIG_NET_CONFIG_MY_IPV4_GW=\"198.51.100.1\" -DCONFIG_ETH_NATIVE_POSIX_DRV_NAME=\"zeth.2\" -DCONFIG_ETH_NATIVE_POSIX_MAC_ADDR=\"00:00:5e:00:53:02\" -DCONFIG_ETH_NATIVE_POSIX_RANDOM_MAC=n
```

Also if you have firewall enabled in your host, you need to allow traffic between zeth.1, zeth.2 and zeth-br interfaces.

jukkar, 2020-04-15:
This is a bug, it is just a matter of configuring the system to setup interfaces and route traffic between them.

closed 2020-04-19 by overheat

jukkar, 2020-05-08:
FYI, I created #25136 that adds proper documentation for connecting multiple Zephyr instances together.