zephyrproject-rtos/zephyr #24291: The button interrupt enters the spurious handler

jasonedn, 2020-04-11:
**Describe the bug**
The button sample ran on the board nucleo_l476rg, When the button was pressed, it entered the spurious handler instead of the button handler. 

**To Reproduce**
Build the button sample on the board nucleo_l476rg with the command:
west build -p auto -b nucleo_l476rg samples/basic/button

**Expected behavior**
When the blue button is pressed, it entered the spurious handler and stuck there.

**Impact**

**Screenshots or console output**
With the debugger, the address of _sw_isr_tables is 0x08003424 and the offset of GPIO exti_15_10 0s 40*8=0x140, so the exti_15_10 address in the table is 0x08003564. From the data below, the ISR address of exti_15_10 is 0x8002BCF that points to the spurious handler instead of the button handler. 
[image]

However according to the generated isr_tables.c, _sw_isr_table is initialized correctly (offset=40) as below.

struct _isr_table_entry __sw_isr_table _sw_isr_table[82] = {
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x2000104c, (void *)0x8002b2b},
	{(void *)0x2000104c, (void *)0x8002b35},
	{(void *)0x2000104c, (void *)0x8002b3f},
	{(void *)0x2000104c, (void *)0x8002b49},
	{(void *)0x2000104c, (void *)0x8002b53},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x2000104c, (void *)0x8002b5d},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x2000104c, (void *)0x8002b67},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
};

**Environment (please complete the following information):**
Toolchain - Zephyr SDK
 
**Additional context**

Why are the source codes correct but the built binary is not?

jasonedn, 2020-04-11:
This issue occurred in Windows but it worked in Linux platform.

erwango, 2020-04-15:
^^@ABOSTM, can you have a check on your side ?

ABOSTM, 2020-04-15:
Hi @jasonedn,
I tested on my side, and I face no issue:
Host: Windows
Zephir SHA1: ab15602b796455ca284abf7234280e6b0ee5e032
Board: nucleo_l476rg
Log console: 

> Button pressed at 790090416
> Button pressed at 1221538460
> Button pressed at 1221649043

Can you tell us more about your environment: host ? toolchain? ...
Can you also try on the same SHA1 than mine ?

jasonedn, 2020-04-15:
Hi Alexandre,

Thanks for your reply.

I tried the same SHA1 as you but it didn't work either. My environments are
as below.
  OS: Windows 10
  Toolchains: /GNU Tools Arm Embedded/9 2019-q4-major

The project was built successfully with west tool, as seen in the
attachment build.log. The generated isr_tables.c is correct but it was not
built into the image.  the _sw_isr_tables still used the default values, so
it jumped to the default spurious handler when the button was pressed.
However from the logs, it reported

The package name passed to `find_package_handle_standard_args`
  (PythonInterp) does not match the name of the calling package (Zephyr).

 Does this help?

Regards
Jason

C:\Users\USER\zephyrproject\zephyr>west build -p auto -b nucleo_l476rg samples/basic/button
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): C:/Users/USER/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/Users/USER/zephyrproject/zephyr/samples/basic/button
-- Zephyr version: 2.2.99 (C:/Users/USER/zephyrproject/zephyr)
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args`
  (PythonInterp) does not match the name of the calling package (Zephyr).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/Users/USER/zephyrproject/zephyr/cmake/backports/FindPythonInterp.cmake:174 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  C:/Users/USER/zephyrproject/zephyr/cmake/python.cmake:22 (include)
  C:/Users/USER/zephyrproject/zephyr/cmake/app/boilerplate.cmake:115 (include)
  C:/Users/USER/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/Users/USER/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: C:/Python38/python.exe (found suitable version "3.8.2", minimum required is "3.6")
-- Board: nucleo_l476rg
-- Found west: C:/Python38/Scripts/west.exe (found suitable version "0.7.2", minimum required is "0.7.1")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNU Tools Arm Embedded/9 2019-q4-major)
-- Found BOARD.dts: C:/Users/USER/zephyrproject/zephyr/boards/arm/nucleo_l476rg/nucleo_l476rg.dts
-- Generated zephyr.dts: C:/Users/USER/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Users/USER/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Users/USER/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Users/USER/zephyrproject/zephyr/build/zephyr/.config'
Configuration saved to 'C:/Users/USER/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/Users/USER/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- Cache files will be written to: C:/Users/USER/zephyrproject/zephyr/.cache
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/USER/zephyrproject/zephyr/build
[119/124] Linking C executable zephyr\zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       15376 B         1 MB      1.47%
            SRAM:        4360 B        96 KB      4.44%
        IDT_LIST:         120 B         2 KB      5.86%
[124/124] Linking C executable zephyr\zephyr.elf

/* AUTO-GENERATED by gen_isr_tables.py, do not edit! */

#include <toolchain.h>
#include <linker/sections.h>
#include <sw_isr_table.h>
#include <arch/cpu.h>
[... 150 lines cut ...]
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
	{(void *)0x0, (void *)&z_irq_spurious},
};

erwango, 2020-04-17:
@andrewboie, we're not able to reproduce the issue, maybe your knowledge of this part will be more efficient to solve this point?

jasonedn, 2020-04-18:
After I uninstalled Visual Studio Community IDE and  zephyr software from my computer, and then I re-installed zephyr software, it worked properly. It seems that it was caused by some improper settings of the VS software.

closed 2020-04-18 by jasonedn