zephyrproject-rtos/zephyr #24300: tests/net/trickle failed on frdm_k64f and sam_e70_xplained with v1.14 branch.

chen-png, 2020-04-13:
**Describe the bug**
Running tests/net/trickle on frdm_k64f and sam_e70_xplained with v1.14-branch, it will **failed occasionally**

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -p auto  tests/net/trickle
2. west flash
3. See error

**Screenshots or console output**
***** Booting Zephyr OS zephyr-v1.14.1-49-gff6829261918 *****
Running test suite test_tickle

starting test - test_init
PASS - test_init

starting test - test_trickle_create
PASS - test_trickle_create

starting test - test_trickle_start
PASS - test_trickle_start

starting test - test_trickle_1_status
PASS - test_trickle_1_status

starting test - test_trickle_2_status
PASS - test_trickle_2_status

starting test - test_trickle_1_wait
[00:00:00.000,000] <wrn> net_if: You have 1 IPv4 net_if addresses but 2 network                                                                                                              interfaces
[00:00:00.000,000] <wrn> net_if: Consider increasing CONFIG_Trickle 1 0x20000c40                                                                                                              callback called
PASS - test_trickle_1_wait

starting test - test_trickle_2_wait
NET_IF_MAX_IPV4_COUNT value.
[00:00:00.000,000] <wrn> net_if: You have 1 IPv6 net_if addresses but 2 network                                                                                                              interfaces
[00:00:00.000,000] <wrn> net_if: Consider increasing CONFIG_NET_IF_MAX_IPV6_COUN                                                                                                             T value.
Trickle 1 0x20000c40 callback called

    **Assertion failed at  zephyr-branch-1.14-github/zephyr/tests/net/trickle/src/main.c:171: test_trickle_2_wait: (cb_2_called is false)**  

**Trickle 2 no timeout
Trickle 2 0x20000c80 callback called
FAIL - test_trickle_2_wait**

starting test - test_trickle_1_update
PASS - test_trickle_1_update

starting test - test_trickle_2_inc
PASS - test_trickle_2_inc

starting test - test_trickle_1_status
PASS - test_trickle_1_status

starting test - test_trickle_1_wait_long
SKIP - test_trickle_1_wait_long

starting test - test_trickle_stop
PASS - test_trickle_stop

starting test - test_trickle_1_stopped
PASS - test_trickle_1_stopped

Test suite test_tickle failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.10.3
 - Commit ID: ff682926

nashif, 2020-04-14:
@jukkar  can you take a look at that one?

jukkar, 2020-04-15:
Already fixed in 2.0 by #18625
I will backport the fix to 1.14

stephanosio, 2020-07-01:
Already fixed by https://github.com/zephyrproject-rtos/zephyr/pull/24370

PR #24370 (2020-04-17): [Backport v1.14-branch] tests: net: trickle: use separate semaphore for second trickle timer
A second semaphore is used for the second trickle timer, so that if the
first timer expires twice before the second one, the test would still
wait before proceeding to check on cb_2_called.

Fixes #18598.

Backported to v1.14 in order to fix PR #24300
Fixes #24300
commit 4730a9b4 (2020-04-17): tests: net: trickle: use separate semaphore for second trickle timer

A second semaphore is used for the second trickle timer, so that if the
first timer expires twice before the second one, the test would still
wait before proceeding to check on cb_2_called.

Fixes #18598.

Backported to v1.14 in order to fix PR #24300
Fixes #24300
closed 2020-07-01 by stephanosio