zephyrproject-rtos/zephyr #24301: Support for multi core STM32 H7 boards

stevewells20, 2020-04-13:
Problem:
There is no support for multi core STM32 boards in the Nucleo in the H7*7 and H7*5 families; these are quite powerful and offer interesting new possibilities.

I would like to know if this is easily possible to implement support for these board families, as there is already support for both Cortex M7 and M4 STM32 boards.

erwango, 2020-04-15:
Hi @stevewells20,

Adding Dual core support for these soc/boards should not be a complex task.
You first need to add device tree for the targeted SoCs.
Then provide board support based on what was done on stm32h747i_disco.

stevewells20, 2020-04-17:
Thank you for the response! I'll look into it. Are there any resources available for best practices or examples for adding new boards? I'm new to Zephyr, and would like to get my feet wet by diving in.

erwango, 2020-04-17:
@stevewells20, you can look at https://github.com/zephyrproject-rtos/zephyr/pull/23160 for reference.
You don't need to provide full support for the board, but a minimum needs to be added like done in this example.
From what I understand, you want to introduce new SoC, so we'll wait 2 commits:
- new SoC introduction
- board introduction

For the board configuration, dual core support is a bit specific, so you can check how it is done for this board:
https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/arm/stm32h747i_disco

If you feel stuck, don't hesitate to push a draft PR and ask for help: @FRASTM or @ABOSTM

Nukersson, 2020-04-26:
I am also working on support for NUCLEO-H745ZI

PR #24712 (2020-06-19): Add nucleo h745zi
Support for NUCLEO-H745ZI-Q with Cortex-M7 and Cortex-M4 added 

Fixes #24301
Fixes #21399
closed 2020-06-19 by carlescufi