zephyrproject-rtos/zephyr #24306: How to set up native posix board to allow connections to the Internet?

overheat, 2020-04-13:
Hi,

[Socket HTTP GET Example](https://docs.zephyrproject.org/latest/samples/net/sockets/http_get/README.html#requirements) require connections to the Internet. But how to do for native posix board? :)

Thanks!

jukkar, 2020-04-13:
See https://docs.zephyrproject.org/latest/guides/networking/qemu_setup.html#setting-up-zephyr-and-nat-masquerading-on-host-to-access-internet for details. That guide talks about QEMU but the same things apply to native-posix.

awsa2ron, 2020-04-16:
Hi,
As the "set the CONFIG_NET_CONFIG_MY_IPV4_GW option to the IP address of the gateway."
`the IP address of the gateway` is my physical router IP address, which is 192.168.0.1? or my zeth interface IP address, which is 192.0.2.2

Thanks.

overheat, 2020-04-16:
> See https://docs.zephyrproject.org/latest/guides/networking/qemu_setup.html#setting-up-zephyr-and-nat-masquerading-on-host-to-access-internet for details. That guide talks about QEMU but the same things apply to native-posix.

Hi @jukkar,
As the "set the CONFIG_NET_CONFIG_MY_IPV4_GW option to the IP address of the gateway."
`the IP address of the gateway` is my physical router IP address, which is 192.168.0.1? or my zeth interface IP address, which is 192.0.2.2

After setting `CONFIG_NET_CONFIG_MY_IPV4_GW` to my zeth interface IP address(192.0.2.2). I can ping 192.168.5.141(host IP, my PC) successfully, but ping 192.168.0.1(router IP) failed.
How to bridge(NAT) to my wlan network, which is 192.168.0.0/20, to access Internet?

Thanks.

jukkar, 2020-04-16:
The GW address is suppose to be your network gateway address, so 192.168.0.1 in this case.

overheat, 2020-04-17:
@jukkar 
I have setup zephyr as you suggested, GW=192.168.0.1, iptables and so on.
Ping my gateway and grab zeth interface traffic on Linux host using wireshark, like this:
[image]

Still not work. Any suggestion?
BTW, iptables like this:
```
$ sudo iptables -t nat -L

...
Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  bogon/16             anywhere            
MASQUERADE  all  --  192.0.2.1            anywhere            
MASQUERADE  all  --  192.0.2.2            anywhere  
```
Thanks.

jukkar, 2020-04-17:
> And zephyr native posix board arp got same MAC address for both 192.0.2.2 and 192.168.0.1.
> 
> ```
> uart:~$ net arp
>      Interface  Link              Address
> [ 0] 0x56600dc0 00:00:5E:00:53:FF 192.168.0.1
> [ 1] 0x56600dc0 00:00:5E:00:53:FF 192.0.2.2
> ```

That does not look right. You have something wrong in your host setup but it is very difficult to debug this remotely via github issue.

overheat, 2020-04-19:
@jukkar 
It works after I disable the firewall on my Kali Linux by `ufw disable`, looks the ufw disabled the ip forwording default. alternately, change the config of ufw to allow ip forwarding should works.
BTW, the gate way is "192.0.2.2"

closed 2020-04-19 by overheat