zephyrproject-rtos/zephyr #24311: LPN not receiving any message from Friend node after LPN device reset

rhardik, 2020-04-13:
**Describe the bug**

I've added a Generic On-Off model in mesh_shell code to publish a message from Friend node to LPN node. it's working fine after provision and configuration 

Now if I reset the LPN node then  friendship get's established again but it doesn't receive any message from friend.

**To Reproduce**
Steps to reproduce the behavior:
1. git clone https://github.com/rhardik/reproduce-zephyr-issues.git
2. cd lpn-node-reset
3. compile friend and lpn code and flash the code to device
4. open serial terminal and press enter. you will see a shell  with $uart
5. Type command "mesh init" and then "mesh pb-gatt on" (for both device)
6. Open nRF mesh android app. Provision and then  
             - configure "generic on-off client" on Friend node  with publish address 0xc000
             - configure "generic on-ff server: LPN node with subscribe address 0xc000
7. Type command "mesh lpn on" in LPN terminal and "mesh friend on" in Friend terminal.
7. Now type command "mesh hello" on Friend terminal
   - After few seconds you should see "HHHHEEELLLLOOOO" string on LPN terminal.
9. Now reset LPN device then 
       - type command  "mesh init" and "mesh lpn on" in LPN terminal and 
       - type "mesh hello" command in Friend terminal 
and now it should print "HHHHEEELLLLOOOO" string on LPN terminal but it won't.

**Expected behavior**
After device reset also it should work.

**Impact**
showstopper

**Screenshots or console output**
Message sent by Friend node (working)
------------------------------
\`\`\`
:~$ mesh hello                                                                                                
[00:28:08.686,004] <dbg> bt_mesh_friend.bt_mesh_friend_match: LPN 0x0003 matched address 0xc000                   
[00:28:08.686,035] <dbg> bt_mesh_friend.bt_mesh_friend_enqueue_tx: net_idx 0x0000 dst 0xc000 src 0x0002           
[00:28:08.686,065] <dbg> bt_mesh_friend.friend_lpn_enqueue_tx: LPN 0x0003                                         
[00:28:08.686,096] <dbg> bt_mesh_friend.enqueue_friend_pdu: type 0                                                
[00:28:08.686,096] <dbg> bt_mesh_friend.friend_lpn_enqueue_tx: Queued message for LPN 0x0003                      
[00:28:09.059,539] <dbg> bt_mesh_friend.bt_mesh_friend_match: LPN 0x0003 matched address 0xc000                   
[00:28:09.059,539] <dbg> bt_mesh_friend.bt_mesh_friend_enqueue_tx: net_idx 0x0000 dst 0xc000 src 0x0002           
[00:28:09.059,570] <dbg> bt_mesh_friend.friend_lpn_enqueue_tx: LPN 0x0003                                         
[00:28:09.059,600] <dbg> bt_mesh_friend.enqueue_friend_pdu: type 0                                                
[00:28:09.059,600] <dbg> bt_mesh_friend.friend_lpn_enqueue_tx: Queued message for LPN 0x0003 
\`\`\`

Message received by LPN node
------------------------------
\`\`\`
uart:~$
HHHHEEELLLLOOOO                                                                                                   
[00:28:08.183,135] <dbg> bt_mesh_lpn.lpn_timeout: state: established                                              
[00:28:08.183,166] <dbg> bt_mesh_lpn.send_friend_poll: lpn->sent_req 0x00                                         
[00:28:08.184,356] <dbg> bt_mesh_lpn.req_sent: req 0x01 duration 90 err 0 state established                       
[00:28:08.184,387] <dbg> bt_mesh_lpn.lpn_set_state: established -> recv delay                                     
[00:28:08.274,475] <dbg> bt_mesh_lpn.lpn_timeout: state: recv delay                                               
[00:28:08.276,123] <dbg> bt_mesh_lpn.lpn_set_state: recv delay -> wait update                                     
[00:28:08.358,062] <dbg> bt_mesh_lpn.friend_response_received: lpn->sent_req 0x01                                 
[00:28:08.358,367] <dbg> bt_mesh_lpn.lpn_set_state: wait update -> established                                    
[00:28:08.358,367] <dbg> bt_mesh_lpn.bt_mesh_lpn_msg_received: Requesting more messages from Friend               
[00:28:08.358,367] <dbg> bt_mesh_lpn.send_friend_poll: lpn->sent_req 0x00                                         
[00:28:08.359,619] <dbg> bt_mesh_lpn.req_sent: req 0x01 duration 90 err 0 state established                       
[00:28:08.359,619] <dbg> bt_mesh_lpn.lpn_set_state: established -> recv delay                   
\`\`\`

Friend side logs when reset LPN
-----------------------------------------------------
\`\`\`
36:27.304,718] <dbg> bt_mesh_friend.friend_timeout: lpn 0x0003 send_last 0 last 0x20006584                    
[00:36:27.304,748] <wrn> bt_mesh_friend: Friendship lost with 0x0003                                              
[00:36:27.304,748] <dbg> bt_mesh_friend.friend_clear: LPN 0x0003  
\`\`\`

Friend logs when sending message to LPN (after LPN reboots)
--------------------------------------------------------------------------------------------
\`\`\`
uart:~$ mesh hello                                                                                                
[00:38:11.604,278] <dbg> bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0xc000                  
[00:38:11.977,630] <dbg> bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0xc000                  
[00:38:22.377,380] <dbg> bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0x0002                  
[00:38:22.384,185] <dbg> bt_mesh_friend.bt_mesh_friend_find: net_idx 0x0000 lpn_addr 0x0003                       
[00:38:22.384,216] <dbg> bt_mesh_friend.bt_mesh_friend_poll: msg->fsn 0 frnd->fsn 1                               
[00:38:22.384,216] <dbg> bt_mesh_friend.friend_recv_delay: Waiting RecvDelay of 151 ms                            
[00:38:22.384,246] <dbg> bt_mesh_friend.encode_update: lpn 0x0003 md 0x00                                         
[00:38:22.384,246] <dbg> bt_mesh_friend.encode_friend_ctl: LPN 0x0003                                             
[00:38:22.384,307] <dbg> bt_mesh_friend.bt_mesh_friend_poll: Enqueued Friend Update to empty queue                
[00:38:22.535,278] <dbg> bt_mesh_friend.friend_timeout: lpn 0x0003 send_last 0 last 0x00000000                    
[00:38:22.535,308] <dbg> bt_mesh_friend.update_overwrite: Update Previous Friend Update MD 0x00 -> 0x00           
[00:38:22.535,430] <dbg> bt_mesh_friend.friend_timeout: Sending buf 0x20006584 from Friend Queue of LPN 0x0003    
[00:38:22.537,567] <dbg> bt_mesh_friend.buf_send_start: err 0                                                     
[00:38:22.598,693] <dbg> bt_mesh_friend.buf_send_end: err 0 
\`\`\`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA: a3fae2f153587c552623ad4940eb6e058aeec78f

LingaoMeng, 2020-04-13:
well, it’s not bug, if you read log from friend node side `[00:38:11.604,278] bt_mesh_friend.bt_mesh_friend_match: No matching LPN for address 0xc000 `, mean that lpn node not subscribe 0xc000 address. So lpn node not receive such message after reset. 

Next: you should add cmd to subscribe 0xc000 when established friendships.

LingaoMeng, 2020-04-13:
well, we should change currently implementation;

Firstly: call lpn group add (mod subscribe address )when recovered from flash.

Secondly: call lpn group add 0xffff, when provisioned completely.

Lastly: call lpn group del, when local node reset or unsubscribe.

Can you submit pr to fix this issuer.

PR #24327 (2020-04-19): Bluetooth: Mesh: Fix lpn groups add when node reset
Add lpn groups when model sub restore after reset.

Add auto sub `ALL_NODE_ADDR` when provisioned.

Remove all groups when node unprovisoned.

Fixes: #24311
Fixes: #24009
commit 49e4f754 (2020-04-19): Bluetooth: Mesh: Fix add model group address to sub list

When low power node reset, should add group address to
friend subscription list.

Fixes: #24311
closed 2020-04-19 by jhedberg