zephyrproject-rtos/zephyr #24313: issues with nrfx uarte hw async validation and behavior

pabigot, 2020-04-13:
When using `nordic,nrf-uarte` the following configuration:
```
CONFIG_UART_ASYNC_API=y
CONFIG_UART_0_NRF_HW_ASYNC=y
```
ends up referencing `CONFIG_UART_0_NRF_HW_ASYNC_TIMER` which by default is empty, so https://github.com/zephyrproject-rtos/zephyr/blob/8b6930ebef7d5945432bd2b3caf452b7f7587a9e/drivers/serial/uart_nrfx_uarte.c#L1394 produces a cascading build error starting with:
```
/mnt/nordic/zp/modules/hal/nordic/nrfx/drivers/include/nrfx_timer.h:62:39:
  error: 'NRF_TIMER' undeclared here (not in a function); did you mean 'NRF_TIMER2'?
   62 |     .p_reg            = NRFX_CONCAT_2(NRF_TIMER, id),             \
      |                                       ^~~~~~~~~
```

It would also be nice if the documentation made it more clear that (a) UARTE can hold received data for an unbounded time without signalling `RX_RDY` if `NRF_HW_ASYNC` isn't enabled, and (b) enabling `NRF_HW_ASYNC` requires that an associated timer be identified.

It's not clear to me whether point (a) is a bug or a feature of using UARTE in the default configuration that lacks hardware RX counting.  So I've tagged this with both labels; please remove one.

Tooling to allow the user to identify a candidate timer without having to guess which ones are in use would also be welcome.

(Point (a) can be reproduced with an async setup where a burst of data exceeding the buffer size is received.

```
# Two 8-char buffers are supplied to UARTE.
0:00:00.012 [1]: last char 00 '?'
# This sequence is supplied in one burst at 2.316: A23456789abc
RX_RDY Buf 0 ofs 0 len 8: 'A2345678'
0:00:02.316 [75504]: last char 38 '8'
RX_RDY Buf 1 ofs 0 len 1: '9'
0:00:02.918 [95236]: last char 39 '9'
# User types a single character: x
RX_RDY Buf 1 ofs 1 len 1: 'a'
0:00:10.218 [6779]: last char 61 'a'
# User types a single character: x
RX_RDY Buf 1 ofs 2 len 1: 'b'
0:00:11.619 [52657]: last char 62 'b'
# User types a single character: x
RX_RDY Buf 1 ofs 3 len 1: 'c'
0:00:13.119 [101812]: last char 63 'c'
# User types a single character: x
RX_RDY Buf 1 ofs 4 len 4: 'xxxx'
0:00:15.122 [167453]: last char 78 'x'
# Now things are back in sync
```

The first buffer fills, then the second one is partly filled, but the driver only notifies of the first character.  More data is dribbled out at the rate of new data until the buffer fills and it's finally flushed.  This isn't exactly the "lost data" mentioned in the Kconfig documentation, because eventually you'll get everything, but it's also not correct behavior.)

pabigot, 2020-04-13:
This behavior occurs only when there is no hardware flow control.

pabigot, 2020-05-11:
The bug part of this appears to be resolved by #24209 so I'm dropping that, leaving this as an enhancement to provide better diagnostics when `CONFIG_UART_0_NRF_HW_ASYNC=y` is provided without a companion RTC assignment like `CONFIG_UART_0_NRF_HW_ASYNC_TIMER=4`.

kartben, 2026-10-30:
The build error was fixed by #24209, and the UART_x_NRF_HW_ASYNC_TIMER option is now deprecated in favor of the devicetree timer property. Closing as obsolete.

closed 2026-10-30 by kartben