zephyrproject-rtos/zephyr #24318: Postpone driver initialization

chris-makaio, 2020-04-13:
## Introduction

Currently drivers are always started during the kernel boot process. I suggest to add a mechanism to disable autostart of specific drivers.

### Problem description

In our use case some drivers are only optional. Think of a fully packed board with tons of sensors and you want a single firmware. In worst case a lot of drivers are started although the specific device might dynamically load config data from the cloud which configures the device to only use a single sensor.

### Proposed change

Allow to set drivers to „manual“ startup mode to let application logic decide if and when to start them

### Proposed change (Detailed)

I know this will possibly have an big impact to the kernel. So as a naive idea I personally thought of using a dedicated init priority (eg 255) to let the kernel know to not call the init fn automatically. That way it‘s not required to change the device drivers but rather have a single change in device.c

carlescufi, 2020-04-21:
CC @tbursztyka @mbolivar-nordic @pabigot

pabigot, 2020-04-21:
Relates to #19448

pabigot, 2020-07-07:
Closing this as a duplicate of #19448 which proposes support for the devicetree standard way of identifying devices as not started at boot.  If there are subtleties of the requirements here please ensure 19448 has enough information to identify them.

closed 2020-07-07 by pabigot