zephyrproject-rtos/zephyr #24339: arm_gic_irq_set_priority - temporary variable overflow

stradebianche, 2020-04-14:
**Bug description**

[Bug location file, starts at line 55](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/interrupt_controller/intc_gic.c)

Issue:
- In function **arm_gic_irq_set_priority** there is a temporary unsigned 8-bit variable _val_. 
- On the later stage _int_off_ variable is calculated for a particular interrupt we want to setup, i.e. for interrupt number 77 (which is TTC3 interrupt on ZynqMP) this values is 26. The equation for _int_off_ comes from GICv2 documentation (as I assume), where corresponding bits offset is calculated from given formula (see references).

- This formula however seems not to be valid for Zephyr approach of 8-bit system reads and writes, as in the next lines we are reading only a >part< of the ICFGR register and then locally modify its content with:
`val &= ~(GICD_ICFGR_MASK << int_off);`
here, the 8-bit _val_ variable **will not be affected**, because we are shifting GICD_ICFGR_MASK left by 26 bits, which is outside of the scope.

- Aside to that, why not use 32-bit versions of read/write functions as they tend to be more reliable (in terms of aligned memory read)?

**Proposed solution**
My proposed solution is to stick to 32-bit read/write system functions and modify size of temporary variable:
```
static arm_gic_irq_set_priority(
		unsigned int irq, unsigned int prio, u32_t flags)
{
	unsigned int int_grp, int_off;
	u32_t val;

	/* Set priority */
	sys_write8(prio & 0xff, GICD_IPRIORITYRn + irq); // I would update this to 32-bit version also

	/* Set interrupt type */
	int_grp = (irq / 16) * 4; // This line is modified according to GICv2 docs
	int_off = (irq % 16) * 2;

	val = sys_read32(GICD_ICFGRn + int_grp);
	val &= ~(GICD_ICFGR_MASK << int_off);
	if (flags & IRQ_TYPE_EDGE) {
		val |= (GICD_ICFGR_TYPE << int_off);
	}

	sys_write32(val, GICD_ICFGRn + int_grp);
}
```

**References**
 -  [The GICv2 ds I am using:](https://static.docs.arm.com/ihi0048/b/IHI0048B_b_gic_architecture_specification.pdf?_ga=2.166146942.113240462.1586350270-16233238.1550224078)   pages 109 and 110.

stephanosio, 2020-04-15:
Thanks for reporting this.

This is indeed a bug; as you suggest the `val` should be `u32_t` and we should `sys_{read,write}32` to `GICD_ICFGRn` since these registers are not byte-accessible (cc @bbolen @wjliang).

As for `GICD_IPRIORITYRn`, the documentation specifically states that "these registers are byte-accessible" (see 4.3.11), so we need not worry about unaligned accesses.

@stradebianche would you be willing to create a pull request for this since you have already provided reasonable a solution above? or would you prefer me to fix this?

stephanosio, 2020-04-15:
cc @carlocaione This might be of interest to you as well, since `qemu_cortex_a53` uses GIC.

stradebianche, 2020-04-15:
@stephanosio you are correct regarding `GICD_ICFGRn`, I must have missed that.
I would like to create a pull request to fix this, so I will proceed with that.

PR #24454 (2020-04-20): interrupt_controller: gic: Fix ICFGRn access and config
Fixing ICFGRn register access with `sys_{read,write}32`
since this register is not byte-accessible.
Type of `val` changed to u32 to match reg width.

Fixes #24339

Supersedes #24422
commit c0be43e9 (2020-04-20): interrupt_controller: gic: Fix ICFGRn access and config

Fixing ICFGRn register access with `sys_{read,write}32`
since this register is not byte-accessible.
Type of `val` changed to u32 to match reg width.

Fixes #24339

Supersedes #24422
closed 2020-04-20 by carlescufi