zephyrproject-rtos/zephyr #24340: #24308 Broke python3 interpreter selection

pfalcon, 2020-04-14:
**Describe the bug**
https://github.com/zephyrproject-rtos/zephyr/pull/24308 broke selection of the Python3 interpreter. Instead of using system (and user's) default "python3" interpreter, it picks up random binaries in uncalled-for manner.

**To Reproduce**
Since #24308 was merged, any attempt to build any Zephyr application leads to error below.
Specific steps to reproduce the behavior:
1. cd samples/net/sockets/big_http_download
2. `west build -b frdm_k64f -- -DOVERLAY_CONFIG=overlay-ci.conf`
3. See error
~~~
$ west build -b frdm_k64f -- -DOVERLAY_CONFIG=overlay-ci.conf
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/samples/net/sockets/big_http_download
-- Zephyr version: 2.2.99 (/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable version "3.8.2", minimum required is "3.6") found components:  Interpreter 
-- Board: frdm_k64f
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'west'
CMake Error at /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/host-tools.cmake:26 (message):
  Unable to import west.version from '/usr/bin/python3.8'
Call Stack (most recent call first):
  /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/cmake/app/boilerplate.cmake:483 (include)
  /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/pfalcon/.local/bin/cmake -B/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr/samples/net/sockets/big_http_download/build -S/mnt/hdd/projects-3rdparty/Embedded/Zephyr/zephyr/samples/net/sockets/big_http_download -GNinja -DBOARD=frdm_k64f -DOVERLAY_CONFIG=overlay-ci.conf
~~~

This happens because:
1. I have a system-default Python interpreter installed as `python3` executable, per the best practices. I'm running Ubuntu 18.04, so this happens to be Pyhon3.6.
2. As a regular Python user, I have many other Python versions installed, e.g. Python3.5, Python3.7, Python3.8, Python3.9, Python-git, etc. etc.

Per the common sense and the best practices, I expect any software which requires Python3 to use system-default executable (`python3`), unless I explicitly used a different Python interpreter to run a specific software. 

**Expected behavior**
Zephyr build system should not try to oversmart user and select arbitrary binaries to run behind user's back. For as long as system-default `python3` interpreter exists, it should use that, unless explicitly asked by a user to do otherwise (via command-line arg or environment var).

**Impact**
This issue breaks development process for any user who may have a newer version of Python installed for reasons not related to Zephyr.

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 18.04
 - Reproduced with master 4215968b74666adedc9c0d643b38d0c874319512

GeorgeGkinis, 2020-04-14:
I am also confronted with this issue.
Any workarounds?

Can I specify the python version to be used within my CMakeLists.txt?

pfalcon, 2020-04-14:
I currently have:

~~~~
--- a/cmake/python.cmake
+++ b/cmake/python.cmake
@@ -9,5 +9,7 @@ if (WIN32)
   set(ENV{PYTHONIOENCODING} "utf-8")
 endif()
 
+#set(Python3_FIND_STRATEGY "LOCATION")
+set(Python3_EXECUTABLE "python3")
 find_package(Python3 3.6 REQUIRED)
 set(PYTHON_EXECUTABLE ${Python3_EXECUTABLE})
~~~

This shows what worked and what not for me.

GeorgeGkinis, 2020-04-14:
Thx,

In my case :
found python version used by west : 
```
pip3 install west
Requirement already satisfied: west in c:\python38\lib\site-packages (0.7.2)
Requirement already satisfied: packaging in c:\python38\lib\site-packages (from west) (20.3)
Requirement already satisfied: setuptools in c:\python38\lib\site-packages (from west) (41.2.0)
Requirement already satisfied: colorama in c:\python38\lib\site-packages (from west) (0.4.3)
Requirement already satisfied: pykwalify in c:\python38\lib\site-packages (from west) (1.7.0)
Requirement already satisfied: configobj in c:\python38\lib\site-packages (from west) (5.0.6)
Requirement already satisfied: PyYAML>=5.1 in c:\python38\lib\site-packages (from west) (5.3)
Requirement already satisfied: pyparsing>=2.0.2 in c:\python38\lib\site-packages (from packaging->west) (2.4.6)
Requirement already satisfied: six in c:\python38\lib\site-packages (from packaging->west) (1.14.0)
Requirement already satisfied: python-dateutil>=2.4.2 in c:\python38\lib\site-packages (from pykwalify->west) (2.8.1)
Requirement already satisfied: docopt>=0.6.2 in c:\python38\lib\site-packages (from pykwalify->west) (0.6.2)

```
then as you suggested : 
```
#set(PYTHON_EXECUTABLE ${Python3_EXECUTABLE})
set(Python3_EXECUTABLE "C:/Python38/python.exe")
find_package(Python3 3.6 REQUIRED)
set(PYTHON_EXECUTABLE ${Python3_EXECUTABLE})
```
I can go to work now :)

tejlmand, 2020-04-14:
> Zephyr build system should not try to oversmart user and select arbitrary binaries to run behind user's back. 

This is not Zephyr trying to oversmart users, but rather CMake itself.

But users has different preferences, some prefers that CMake tries to always locate latest.
Some might want CMake FindPython3 to prefer `/usr/bin/python3` over any `/usr/bin/python3.x` as long as the former is satisfying the stated requirement, other prefer that CMake always selects the latest version (which is the current CMake behaviour).

But I also see the preference for being able to control this a little better.
I will try to see if better control can be accomplished without introducing downstream patches to CMake FindPython3, as doing so tends to lead to other issues.

Have you tried to see if any flags supported natively in CMake, helps to reach the desired effect ?
For example, using the `Python3_ROOT_DIR` flag ?

tejlmand, 2020-04-14:
@GeorgeGkinis according to CMake FindPython3, then on windows, it should use the registry setting.
as per: https://cmake.org/cmake/help/latest/module/FindPython3.html#hints
> FIRST: Try to use registry before environment variables. This is the default.

So you could try to check if the Python 3.8 is correctly registered in the registration database in Windows., should be under:
`[HKEY_CURRENT_USER\SOFTWARE\Python\PythonCore\` or
`[HKEY_LOCAL_MACHINE\SOFTWARE\Python\PythonCore\`

pfalcon, 2020-04-14:
> This is not Zephyr trying to oversmart users, but rather CMake itself.

That's absolutely true, and I don't even want to start on how some people were unhappy with appearance of CMake in Zephyr, and are unhappy with any issue it brings ;-). Let me stay on the positive side and thank CMake maintainers in Zephyr for shielding users from CMake and ask for keeping up this great work. In this case, CMake bled thru...

> Have you tried to see if any flags supported natively in CMake, helps to reach the desired effect ?
For example, using the Python3_ROOT_DIR flag ?

I did a quick google-up and reading on that stuff, tried `set(Python3_FIND_STRATEGY "LOCATION")` which didn't have visible effect for me (3.8 was still selected), then `set(Python3_EXECUTABLE "python3")` worked.

Of course, local-patching cmake/python.cmake in Zephyr tree isn't scalable solution. My suggestion for user-friendly solution would be: issue an equivalent of `which python3`, if that succeeds, execute `set(Python3_EXECUTABLE "python3")`. Only if python3 was not found (Windows systems, etc.) then let `find_package(Python3 3.6 REQUIRED)` do its full deeds.

tejlmand, 2020-04-14:
> are unhappy with any issue it brings ;-)
yeah, unfortunately nothing is perfect, but I'm doing my best to handle things that bothers people.

> I did a quick google-up and reading on that stuff, tried set(Python3_FIND_STRATEGY "LOCATION") which didn't have visible effect for me (3.8 was still selected), then set(Python3_EXECUTABLE "python3") worked.

yeah, unfortunately, this solution is not bulletproof either.
Tried to mimic your setup locally, by having `/usr/bin/python3 --> /usr/bin/python3.6` and then also having `/usr/bin/python3.7` and with CMake 3.14, I get Python 3.6 selected, but CMake 3.16 and 3.17, I constantly get Python 3.7 selected.

And then I haven't even got started on testing on systems, where `/usr/bin/python` is also pointing to a 3.x python. After the deprecation of Python 2.7, then we must expect more systems to have `/usr/bin/python --> /usr/bin/python3.x`.

So what should be default when we face:
`/usr/bin/python --> /usr/bin/python3.x` and `/usr/bin/python3 --> /usr/bin/python3.y`

I agree, it should definitely be possible for users to easily prefer `/usr/bin/python[3]` over `/usr/bin/python3.x`.
Once that is possible, then we can discuss the default behavior.

So still looking into the best way to fix this.

PR #24364 (2020-04-16): cmake: FindPython3: Adjust python3 program search
This PR fixes #24340.

The `find_package(Python3)` will default select the highest installed Python3 version, unless exact is specified. Although it is possible to use CMake flags to control the behavior, then the discussion in #24340 indicates that CMake version 3.14 behaves differently than CMake 3.16 and 3.17.

The desired approach of whether Zephyr should follow CMake `find_package(Python3)` default behavior or another is open for debate.

But this PR provides a consistent mechanism across CMake 3.13-3.17 on how to control the behavior.

Th [... cut]
commit 94de3e9f (2020-04-16): cmake: FindPython3: Adjust python3 program search

Fixes: #24340

Using find_package(Python3 3.6) will select the highest available
python3.

This means that in case user has:
/usr/bin/python  --> python2.7
/usr/bin/python3 --> python3.6
/usr/bin/python3.6
/usr/bin/python3.8

then CMake will choose python 3.8.
This commit changes that behavior, so that in the above scenario, then
Python 3.6 will be preferred.

It also adds the following, python will be preferred over python3, if
both meets the minimum requirement.
For example:
/usr/bin/python  --> python3.6
/usr/bin/python3 --> python3.7
then  [... cut]
closed 2020-04-16 by nashif