zephyrproject-rtos/zephyr #24356: MCUboot broken with current zephyr master

henrikbrixandersen, 2020-04-14:
**Describe the bug**
Compiling and running the standard MCUboot sample (`bootloader/mcuboot/samples/zephyr`) on e.g. `frdm_k64f` is broken with current Zephyr master. It works out-of-the-box with tag `v2.2.0`. The error is not restricted to the `frdm_k64f`.

`** Booting Zephyr OS build zephyr-v2.2.0-1438-g3e0d4be27121 ***
[00:00:00.005,000] <inf> mcuboot: Starting bootloader
[00:00:00.011,000] <err> mcuboot: Flash device FLASH_CTRL not found`

**To Reproduce**
Steps to reproduce the behavior:
1. cd bootloader/mcuboot/samples/zephyr
2. make BOARD=frdm_k64f all
3. make flash_full
4. See error on console

**Expected behavior**
The MCUboot sample should boot and show a hello, world message.

**Impact**
MCUboot does not work with Zephyr master.

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04 LTS
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: zephyr-v2.2.0-1438-g3e0d4be27121
**Additional context**
Comments on Slack suggests other flash samples may be affected too.

henrikbrixandersen, 2020-04-14:
Commit 608dba486eebef39317b4e680be36abef297a0cd seems to be the culprit. It changes the label used for the flash controllers, thus applications using `device_get_binding(DT_FLASH_DEV_NAME)` now get a `NULL` return value.

henrikbrixandersen, 2020-04-14:
The `tests/subsys/storage/flash_map` test also fails on `frdm_k64f` due to this. If `608dba486eebef39317b4e680be36abef297a0cd` is reverted, the test succeeds. Note that the `frdm_k64f` needs `CONFIG_MPU_ALLOW_FLASH_WRITE=y` for this test to pass regardless of the above mentioned commit.

mbolivar-nordic, 2020-04-14:
@brix thank you for the detailed report.

The following workflow still works for me on master (as of 1f3c014a1ad302076223e5c1ec3a4892db378af9):

```
$ cd ~/zephyrproject
$ west build -b nrf52840dk_nrf52840 -s bootloader/mcuboot/boot/zephyr/ -d build-mcuboot
$ west flash -d build-mcuboot
$ cd zephyr
$ west build -b nrf52840dk_nrf52840 -s samples/hello_world/  -- -DCONFIG_BOOTLOADER_MCUBOOT=y
$ west sign -t imgtool -- --key ../bootloader/mcuboot/root-rsa-2048.pem
$ west flash --hex-file /home/mbolivar/zp/zephyr/build/zephyr/zephyr.signed.hex
```

So I guess MCUboot isn't what's broken, but its sample application. Looking at that now.

mbolivar-nordic, 2020-04-14:
> So I guess MCUboot isn't what's broken, but its sample application. Looking at that now.

From ~/zephyrproject/bootloader/mcuboot/samples/zephyr/hello-world, this works for me also:

```
$ west build -b nrf52840dk_nrf52840
$ west sign -t imgtool -- --key ../../../root-rsa-2048.pem
$ west flash --hex-file /home/mbolivar/zp/bootloader/mcuboot/samples/zephyr/hello-world/build/zephyr/zephyr.signed.hex
```

So perhaps the Makefile provided in ~/zephyrproject/bootloader/mcuboot/samples/zephyr has some issue. Let's see.

galak, 2020-04-15:
Fixed by #24247

closed 2020-04-15 by galak