zephyrproject-rtos/zephyr #24357: NVS sample on STM32F4 fails even if the dts definition is correct

akofoed, 2020-04-14:
**Describe the bug**
Following @Laczen 's request and our investigation in #24237, tests on the STM32 show that the NVS driver in the current master fails even if the DTS is correct. The NVS driver works on zephyr-v2.2.0. I have no idea as to why.

**To Reproduce**
1) Correct the DTS storage_partition with this (taking the area to start at 1M):
```
               /* storage: 64KB for settings */
		storage_partition: partition@100000 {
			label = "storage";
			reg = <0x00100000 0x00010000>;
		};
```
2) Run the NVS sample on the nucleo_f429zi
3) compile and run
4) The following output occurs (I have added nvs_write rc line to main.c here):
```
Port /dev/ttyACM0, 18:17:13

Press CTRL-A Z for help on special keys

*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
nvs_write rc: -5
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.007,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: 
 Error, no Reboot counter
*** Booting Zephyr OS build zephyr-v2.2.0-1102-gf57ec57ce557  ***
No address found, adding 192.168.1.1 at id 1
nvs_write rc: -5
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.007,000] <inf> fs_nvs: 3 Sectors of 2048 bytes
[00:00:00.007,000] <inf> fs_nvs: alloc wra: 0, 7f0
[00:00:00.007,000] <inf> fs_nvs: data wra: 0, 0
```

**Expected behavior**
Switch to zephyr-v2.2.0 with the above DTS still running on the nucleo_f429zi and observe the following output:
```
*** Booting Zephyr OS build zephyr-v2.2.0  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.014,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.014,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.014,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: ...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
Id: 3, Reboot_counter: 1
Id: 4, Data: DATA
Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 2 
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3fc0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, a1
Reboot counter history: ...1...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
```

**Impact**
NVS not working on STM32F4

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu)
 - Toolchain:Zephyr SDK
 - Commit SHA: 071a986bb676dc10252f92052ba056cc47717195

**Additional context**
N/A

FRASTM, 2020-04-16:
I also see the error with nucleo_g431 but nucleo_l4r5zi is OK

FRASTM, 2020-04-17:
actually for g431rb, error is :+1: 
`[00:00:00.005,000] <err> flash_stm32: Read range invalid. Offset: 133112, len: 8`
Reading the nvs_ate fails after the first `nvs_flash_ate_rd(fs, *addr, ate); `
where  offset = 0x1f000  (which corresponds to the storage partition)
where *addr = 0x7f0
 because the  /* last ate in sector", do jump to previous sector */
gives an un-mapped memory address 133112 (0x207F8). This is the result of the `*addr += ((fs->sector_count - 1) << ADDR_SECT_SHIFT);`

Laczen, 2020-04-17:
> actually for g431rb, error is :
> `[00:00:00.005,000] <err> flash_stm32: Read range invalid. Offset: 133112, len: 8`
> Reading the nvs_ate fails after the first `nvs_flash_ate_rd(fs, *addr, ate); `
> where offset = 0x1f000 (which corresponds to the storage partition)
> where _addr = 0x7f0 because the /_ last ate in sector", do jump to previous sector */
> gives an un-mapped memory address 133112 (0x207F8). This is the result of the `*addr += ((fs->sector_count - 1) << ADDR_SECT_SHIFT);`

The definition for the g431rb does not allow 3 sectors to be used (only 2 sectors are defined as storage). Either increase the definition to 3 sectors or modify the nvs sample to use only 2 sectors.

akofoed, 2020-04-17:
I have been looking at the changelog to find out what has changed between v2.2.0 and current-master but I can't find anything that explains why the flash-driver gives this -5 error.
I have done a test using flash_area_write() and that too gives a -5 (STM32F429ZI) using the 0x100000 128KB area (current-master).

FRASTM, 2020-04-17:
@Laczen Thanks for your comment, which makes this sample test OK on G431, if I set the  fs.sector_count = 2U;  in the sample main.
BTW, even if it's a sample, this could be linked to a config or a macro.
Regarding the definition to 3 sectors of the G431, does it refer to the page size ?
```		storage_partition: partition@1f000 {```
Because if I try to change the dts to (3 x 1024) 
```		storage_partition: partition@1f000 {
			label = "storage";
			reg = <0x00018000 0x00001800>
```
I can also make the sample pass with 3 sectors, but I get a compil warning : unit address and first address in 'reg' (0x18000) don't match for /soc/flash-controller@40022000/flash@8000000/partitions/partition@1f000

erwango, 2020-04-17:
> I have been looking at the changelog to find out what has changed between v2.2.0 and current-master but I can't find anything that explains why the flash-driver gives this -5 error.

@akofoed, can you rebase on top of latest master. There had been a few hickups on stm32 flash drivers few weeks ago. You might face an issue already solved. You can check `samples/drivers/flash_shell`: If not working you need to rebase.

Laczen, 2020-04-17:
> @Laczen Thanks for your comment, which makes this sample test OK on G431, if I set the fs.sector_count = 2U; in the sample main.
> BTW, even if it's a sample, this could be linked to a config or a macro.
> Regarding the definition to 3 sectors of the G431, does it refer to the page size ?
> ` storage_partition: partition@1f000 {`
> Because if I try to change the dts to (3 x 1024)
> 
> ```
> 			label = "storage";
> 			reg = <0x00018000 0x00001800>
> ```
> 
> I can also make the sample pass with 3 sectors, but I get a compil warning : unit address and first address in 'reg' (0x18000) don't match for /soc/flash-controller@40022000/flash@8000000/partitions/partition@1f000

The sector size should always be a multiple of the pagesize.

akofoed, 2020-04-17:
@Laczen  After updating master just now it works. This is NVS on current master running on nucleo_f429zi and working:

```
Booting Zephyr OS build zephyr-v2.2.0-1378-g071a986bb676  ***
No address found, adding 192.168.1.1 at id 1
No key found, adding it at id 2
No Reboot counter found, adding it at id 3
Id: 4 not found, adding it
Longarray not found, adding it as id 5
[00:00:00.016,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.016,000] <inf> fs_nvs: alloc wra: 0, 3ff0
[00:00:00.016,000] <inf> fs_nvs: data wra: 0, 0
Reboot counter history: ...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
*** Booting Zephyr OS build zephyr-v2.2.0-1378-g071a986bb676  ***
Id: 1, Address: 192.168.1.1
Id: 2, Key: ff fe fd fc fb fa f9 f8 
Id: 3, Reboot_counter: 1
Id: 4, Data: DATA
Id: 5, Longarray: 0 1 2 3 4 5 6 7 8 9 a b c d e f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 2 
[00:00:00.017,000] <inf> fs_nvs: 3 Sectors of 16384 bytes
[00:00:00.017,000] <inf> fs_nvs: alloc wra: 0, 3fc0
[00:00:00.017,000] <inf> fs_nvs: data wra: 0, a1
Reboot counter history: ...1...0
Oldest reboot counter: 0
Rebooting in ...5...4...3...2...1
```

Can we close this or do you need me to do more testing?

akofoed, 2020-04-18:
I was testing the jedec,spi-nor version and my code says it cannot find the driver. So I tried the sample in samples/drivers/spi_flash/ and it too cannot find the driver. I have tested this on both current master and v2.2.0.

To see the error you don't need the spi-nor flash attached to your board. It is the spi-nor driver that it cannot find.
This is my addition to the DTS:
```
&spi1 {
	status = "okay";
	clock-frequency = <10000000>;
	cs-gpios = <&gpioe 11 0>;
	flash1: mx25r6435f@0 {
		compatible = "jedec,spi-nor";
		status = "okay";
		label = "MX25R64";
		spi-max-frequency = <10000000>;
		reg = <0>;
		jedec-id = [c2 28 17];
		size = <67108864>;
		wp-gpios = <&gpioe 9 0>;
		hold-gpios = <&gpioe 10 0>;
	};
};

&flash1 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		nvs_partition: partition@0 {
			label = "nvs-partition";
			reg = <0x00000000 0x00010000>;
		};
	};
};
```
Yes, I am using spi4 (gpioE11->E14) but I get the same error as in the sample that the driver cannot be found.
This is the output using the current master (just updated):
```
*** Booting Zephyr OS build zephyr-v2.2.0-1677-ge7c7f911a9cd  ***

JEDEC SPI-NOR SPI flash testing
==========================
SPI flash driver MX25R64 was not found!
```

Output on v2.2.0:
```
*** Booting Zephyr OS build zephyr-v2.2.0  ***

JEDEC SPI-NOR SPI flash testing
==========================
SPI flash driver MX25R64 was not found!
```
Am I missing something or can this be confirmed?

Laczen, 2020-04-19:
@akofoed, make this a separate question and/or bug report. People will not find this problem in the current issue. As the present issue seems to be solved you can close it.

akofoed, 2020-04-19:
I can confirm that this is working now. Closing issue.

closed 2020-04-19 by akofoed