zephyrproject-rtos/zephyr #24363: build system needs to distinguish between free and paid NSIM versions

andrewboie, 2020-04-14:
We have logic in `cmake/emu/nsim.cmake` to find the `nsimdrv` binary, but the build system currently can't distinguish between the free and commercial versions.

When doing sanitycheck runs with nSIM 2019.09 013_FREE, the build system should be aware that we are using the free version and not try to emulate targets that require paid NSIM. This is at least nsim_hs_smp, and I suspect nsim_sem_normal.

This detection could be something simple like scanning the output of `nsimdrv` with no arguments and looking for the regex `Version.*FREE` or something like that.

Current impact is a great deal of sanitycheck failures, which are not actual failures, when running with the free version of NSIM.

andrewboie, 2020-04-14:
@vonhust @ruuddw @abrodkin FYI

andrewboie, 2020-04-15:
Per Synopsys the root cause of this is nsim_hs_smp can't work under ANY version of nsim.

The problem with nsim_sem_normal is unrelated, the code is simply broken. Filing a separate bug.

abrodkin, 2020-04-15:
Indeed so-called "stand-alone nSIM" (i.e. `nsimdrv` alone w/o MetaWare Debugger or SystemC wrapper) currently doesn't support multi-processor or multi-core configurations (AMP, SMP etc). So until this gets fixed we might want to not use `nsim_hs_smp` for CI.

I'm not sure though if there's a way to just exclude one particular platform/board from any CIs so that it sits in the source tree as a "staging" platform. If keeping such kind of platform in the tree is not desired the only thing we may do is to drop it.

andrewboie, 2020-04-15:
This is not just a CI issue. An end user will see this target in the tree and expect it to work, when it doesn't.

Why is there such resistance to removing this non-functional stuff? It's in the git history, you can put it back if and when it is ever fixed.

abrodkin, 2020-04-15:
It's not a resistance. But being a long-time contributor & user of open source projects I know what is a pain of maintaining off-the-tree stuff. Even now working on QEMU platform for ARC I'm seeing problems rebasing my platform from time to time because some things change across the tree but not for my off-the-tree code. And here're some examples: https://github.com/zephyrproject-rtos/zephyr/commit/c5839f834b30 & https://github.com/zephyrproject-rtos/zephyr/commit/6fd168e9a109.

Thus if it's ever possible I'd prefer to keep code as much as possible in the main tree. So that's very practical thing. And as I mentioned if in Zephyr we don't want staging or WIP stuff in the tree then it's OK rules are rules.

nashif, 2020-04-15:
We just need to fix this target not to use standalone nsim for simulation, so something like that should do:

```diff
diff --git a/boards/arc/nsim/nsim_hs_smp.yaml b/boards/arc/nsim/nsim_hs_smp.yaml
index 6cf0d80d1d..8a12b5fa16 100644
--- a/boards/arc/nsim/nsim_hs_smp.yaml
+++ b/boards/arc/nsim/nsim_hs_smp.yaml
@@ -1,7 +1,7 @@
 identifier: nsim_hs_smp
 name: Multi-core HS nSIM simulator
 type: mcu
-simulation: nsim
+simulation: nsim-smp
 arch: arc
 toolchain:
   - zephyr
```

assuming we will start seeing results of this being run in the appropriate environment on testrail soon.

PR #24532 (2020-04-20): boards: nsim: for smp testing, using mdb
This board require a special setup and does not work with standalone
nsim.

Fixes #24363
commit 56aac64b (2020-04-20): boards: nsim: for smp testing, using mdb

This board require a special setup and does not work with standalone
nsim.

Fixes #24363
closed 2020-04-20 by nashif