zephyrproject-rtos/zephyr #24366: syscall for socketpair(2)

cfriedt, 2020-04-15:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

A typical means of inter-thread communication is to use a [socket pair](https://pubs.opengroup.org/onlinepubs/9699919799/functions/socketpair.html).

Although it may be a slightly heavyweight method of inter-thread communication for Zephyr, it would facilitate porting applications.  

**Describe the solution you'd like**
I would like to see an implementation of [socketpair(2)](https://pubs.opengroup.org/onlinepubs/9699919799/functions/socketpair.html) in Zephyr. 

**Describe alternatives you've considered**

**Additional context**

pfalcon, 2020-04-15:
> A typical means of inter-thread communication is to use a socket pair.

I wouldn't call it a "typical" way.

> Although it may be a slightly heavyweight method of inter-thread communication for Zephyr, it would facilitate porting applications.

Sounds good. Would give even more weight to the proposal if you gave example of application(s) which could be ported using it.

cfriedt, 2020-04-15:
@pfalcon - I often use it as a means of cancelling a worker thread where sockets are used with select(2) or poll(2).  If you add one end of the socketpair to readfds or pollfds, then it will unblock a blocking select or poll.

pfalcon, 2020-04-15:
> I often use it as a means of cancelling a worker thread where sockets are used with select(2) or poll(2). 

Then I'd suggest to look into https://github.com/zephyrproject-rtos/zephyr/pull/22863 , and help to review/test it. eventfd() would be much efficient way to signal events (vs passing data) between threads.

cfriedt, 2020-04-15:
> > I often use it as a means of cancelling a worker thread where sockets are used with select(2) or poll(2).
> 
> Then I'd suggest to look into #22863 , and help to review/test it. eventfd() would be much more efficient way to signal events (vs passing data) between threads.

Interesting. Yeah, I guess eventfd() would satisfy my criteria, although it is 100% Linux specific / not portable to other OS's (AFAIK). The nice thing about socketpair() is that it is POSIX and pulled in with `<sys/socket.h>`, so it Just Works (TM) everywhere.

It seems eventfd also only supports writing `uint64_t` instead of arbitrary data, which is somewhat inflexible.

I do think that a pipe(2) implementation would be nice for Zephyr too..

pfalcon, 2020-04-15:
> I guess eventfd() would satisfy my criteria, although it is 100% Linux specific / not portable to other OS's (AFAIK)

To the best of my knowledge, yes.

> It seems eventfd also only supports writing uint64_t instead of arbitrary data, which is somewhat inflexible.

I'd say the basic function of eventfd is to just get out of poll() call. In such usage, you could (almost) always just fallback to (simple) pipe() call. Passing any data is an extra, advanced feature (well, still achievable with pipe() of course).

> I do think that a pipe(2) implementation would be nice for Zephyr too..

+1

> Personally, I don't think having more options is a bad thing.

I think so too, but:

1. More work for you ;-)
2. If we do it, let's do it: a) without obvious security issues; b) truly POSIX compatible (implementing just a subset of functionality for starters is OK, just that subset should be compatible).
3. It's also more work for maintainers, so p.2 is important.

There was also p.4, but I forgot what it was, and hopefully it won't apply to your PRs ;-).

Thanks!

PR #24813 (2020-05-10): net: socket: syscall for socketpair(2)
Working:
* non-blocking reads / writes
* blocking reads / writes
* send(2) / recv(2) / sendto(2) / recvfrom(2) / sendmsg(2)
* select(2)
* poll(2)

Fixes #24366

```
*** Booting Zephyr OS build zephyr-v2.2.0-2516-g63d10feeb7b8  ***
Running test suite socketpair
===================================================================
starting test - test_socketpair_AF_LOCAL__SOCK_STREAM__0
PASS - test_socketpair_AF_LOCAL__SOCK_STREAM__0
===================================================================
starting test - test_socketpair_AF_UNIX__SOCK_STREAM__0
PAS [... cut]
commit 09f957c4 (2020-05-10): net: socket: syscall for socketpair(2)

Working:

* non-blocking reads / writes
* blocking reads / writes
* send(2) / recv(2) / sendto(2) / recvfrom(2) / sendmsg(2)
* select(2)
* poll(2)

Fixes #24366
closed 2020-05-10 by carlescufi