zephyrproject-rtos/zephyr #24373: NULL-pointer dereferencing in GATT when master connection fails

mtpr-ot, 2020-04-15:
In master role, when a connection fails with _failure to establish connection_ (62), and executing GATT commands in _connected_ callback, the system fails with NULL-pointer dereferencing.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to advertiser in master role
2. In _connected_ callback, call `bt_gatt_discover` and `bt_gatt_exchange_mtu`
3. As part of the disconnection flow, _gatt_mtu_rsp_ is called, with `params->func == NULL`
4. NULL-pointer is dereferenced in _gatt_mtu_rsp_, and system crashes

**Expected behavior**
Correct data passed to function, or NULL-pointer checked and ignored. No system failure.

**Impact**
It seems that this only happens with specific combinations of `bt_gatt` calls, combined with the connection failure.

**Additional context**
Vendor specific configuration and platform used.

carlescufi, 2020-04-15:
Closing since #24354 is merged

closed 2020-04-15 by carlescufi

M1cha, 2022-07-21:
This issue shouldn't have been closed since it was created after the merge, exactly to track backporting it to 2.2.x:
https://github.com/zephyrproject-rtos/zephyr/pull/24354#issuecomment-613906317

This was never done and the bug is still present in 2.2.1. Our company just ran into it this week.