zephyrproject-rtos/zephyr #24382: disco_l475_iot1 not working with samples/net/wifi

nandojve, 2020-04-15:
**Describe the bug**
The ST disco_l475_iot1 is not working with samples/net/wifi when enable DHCPv4

**To Reproduce**
Add below config at prj.conf
_CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_DHCPV4=y_

_west build -b disco_l475_iot1 samples/net/wifi
west flash_

**Expected behavior**
board be capable to have console, connect to wifi network and send a ping.

**Screenshots or console output**
```
[uart:~$ *** Booting Zephyr OS build zephyr-v2.2.0-1408-gacc0e9025d5e  ***
[00:00:00.539,000] <inf> net_config: Initializing network
[00:00:00.539,000] <inf> net_config: Running dhcpv4 client...
uart:~$ \���񆿆�ο��YjلO}R��B���᭟�o�1߻��ﻏ_y������׾����ϝo��q]��5�N�O������������������O�
```

**Expected behavior**
board be capable to have console, connect to wifi network and send a ping.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.2
 - Commit SHA or Version used: mainline

erwango, 2020-04-15:
^^@ABOSTM

mtahirbutt, 2020-04-15:
@nandojve is your IP not managed by WiFi? I suppose CONFIG_NET_DHCPV4=n should be instead of y. For me it works with CONFIG_NET_DHCPV4=n

ABOSTM, 2020-04-16:
Hi @nandojve ,
samples/net/wifi is well working (and specially on disco_l475_iot1, that I tested) for what it is intend to do, as said in the README:
'enabling the Wi-Fi shell module'
You cannot just active few switches to have DHCP up and running. Have a look at "\samples\net\dhcpv4_client", you will see that much more switches need to be activated, and that you need to call more code in your application to start DHCP (see main.c).
Also as you are in wifi (instead of ethernet), before starting dhcp, you will need to connect to a specific wifi network, provide password, ...

nandojve, 2020-04-16:
I'm start to think that board that I received have some defect. On the regular path following all steps on the wifi demo I can connect to network but I can't see any ping return even from my local network. I don't have a default gateway and netmask. When I enabled below flags wrong characters show on the console and I lose control.

`uart:~$ \���񆿆�ο��YjلO}R��B���᭟�o�1߻��ﻏ_y������׾����ϝo��q]��5�N�O������������������O�`

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_DHCPV4=y

What could I doing here on wifi demo enabling above flags that could create such problem with console?

mtahirbutt, 2020-04-16:
can you ping your board from another station? I tried that although did not tried ping from the board.

mtahirbutt, 2020-04-16:
you can also try echo_client from zephyr there is already available Digi Servers available for both UDP and TCP.

ABOSTM, 2020-04-17:
> What could I doing here on wifi demo enabling above flags that could create such problem with console?
My 1st guess would be that DHCP starts but the board is not connected to a wifi Access Point.

erwango, 2020-04-17:
@loicpoulain any idea?

nandojve, 2020-04-17:
Hi folks, yesterday I got communication working using APIs and get a transfer from a  COAP server using my local network without using DNS. I still not getting PING and DNS working with net shell.
It seems that when I simple enable DHCPv4 option shell start to send spurious on console and I lose shell control.

@jukkar , any idea about wrong characters on shell when enable DHCPv4 option.

jukkar, 2020-04-21:
> You cannot just active few switches to have DHCP up and running. Have a look at "\samples\net\dhcpv4_client", you will see that much more switches need to be activated, and that you need to call more code in your application to start DHCP (see main.c).

If one enables the `CONFIG_NET_CONFIG_SETTINGS=y`, then the operations done in `samples/net/dhcpv4_client` are done automatically meaning `net_dhcpv4_start()` is called by the config lib in that case. The DHCPv4 sample is there just to show how one can enable DHCPv4 if the config settings library is not enabled. So for the other network samples, it should be enough to just enable `CONFIG_NET_DHCPV4=y` as the config settings is enabled already for them. If one has an own application, then one might need to call `net_dhcpv4_start()` manually (if the config settings lib is not enabled).

> any idea about wrong characters on shell when enable DHCPv4 option.

I have not seen this kind of garbage output, but I have not used wifi shell much so this could be something related to that shell.

ABOSTM, 2020-04-22:
> If one enables the CONFIG_NET_CONFIG_SETTINGS=y, then the operations done in samples/net/dhcpv4_client are done automatically meaning net_dhcpv4_start() is called by the config lib in that case. 

Ok, so dhcp starts whereas there is not connection to wifi AccessPoint. I suspect garbage comes from there. So maybe it is not good to enable  CONFIG_NET_CONFIG_SETTINGS in this case, but rather in the application, establish first the wifi connection to the Access Point and then only start DHCP.

jukkar, 2020-04-22:
> Ok, so dhcp starts whereas there is not connection to wifi AccessPoint. I suspect garbage comes from there. So maybe it is not good to enable CONFIG_NET_CONFIG_SETTINGS in this case, but rather in the application, establish first the wifi connection to the Access Point and then only start DHCP.

That sounds like a good plan. The net config settings lib tries to wait until there is a connection available, but if it takes long time, then it might timeout etc.

loicpoulain, 2020-04-22:
@erwango, @nandojve,  The WiFi controller is quite autonomous and provides a **TCP/UDP offloading** interface and not a raw 802.11 or ethernet interface. That means you will be only able to create TCP/UDP socket in order to act as client or server. Protocol like ICMP (for ping) can not go through the chip. The chip, however, integrates a DHCP client/server that is started via an AT command (This is something trigged inside the driver) and even a ping capatibility (also via an AT command).

Offloading allows to keep the host quite simple (no IP, TCP or UDP stack requested), e.g. to create an Acces Point, a simple AT command is requested and the chip fully manage DHCP, client connection and authentication, packet rounting, etc...

closed 2020-04-22 by jukkar

erwango, 2020-04-24:
Txs @loicpoulain !

PR #24590 (2020-04-22): Fix net_if_send_data for non-l2 iface 
Some services like DHCHPv4 directly send raw packets to the iface.
This causes issue when the iface does not implement l2, e.g.
because it is a socket offload interface. fix that.

Fixes: #24382