zephyrproject-rtos/zephyr #24409: When the delay parameter of k_delayed_work_submit is K_FOREVER, the system will crash

LuoZhongYao, 2020-04-16:
**Describe the bug**
If the k_delayed_work_submit function is called with the K_FOREVER parameter, the system will error

**To Reproduce**
1. Use any board
2. Call the k_delayed_work_submit function and set the parameter to K_FOREVER, then call k_delayed_work_submit through another k_delayed_work

**Expected behavior**
Calling k_delayed_work_submit with K_FOREVER can directly return an error, or directly return..

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain Zephyr SDK
 - Commit SHA 40ee38a9450598f8ddc384cf891fc55a44ded38f

carlescufi, 2020-04-21:
@andyross we are likely missing a test for this particular usage of delayed work?

andyross, 2020-04-21:
This actually never worked correctly (well, "correctly" -- it's a pathological case after all), and AFAICT the behavior didn't change with the timeout patch.  What actually happens in practice is that the delayed work timeout gets set for -1 (== K_FOREVER) ticks into the future, which means it fires immediately at the next timer interrupt.

I can't reproduce a crash, btw; it just fires normally but unexpectedly.  Maybe your app is hitting an unexpected situation and crashing?

As far as a fix: we can either document this as a "don't do that", or (what I think I'll submit) add a check to z_add_timeout() which noops if a K_FOREVER reaches down that far.

PR #24565 (2020-04-22): kernel/timeout: Check for K_FOREVER in z_add_timeout()
The "forever" token has always been interpreted above z_add_timeout()
(because it's always taken ticks, but K_FOREVER used to be in ms).
But it was discovered that k_delayed_work_submit_to_queue() was never
testing for this and passing a raw K_FOREVER down, where it got
interpreted as a negative timeout and caused it to fire at the next
tick.

Now that we actually see the original k_timeout_t here, we might as
well check it locally and do the correct thing (that is, nothing) if
asked to schedule a timeout that will never fire.

Fixes #2440 [... cut]
commit 12bd1870 (2020-04-22): kernel/timeout: Check for K_FOREVER in z_add_timeout()

The "forever" token has always been interpreted above z_add_timeout()
(because it's always taken ticks, but K_FOREVER used to be in ms).
But it was discovered that k_delayed_work_submit_to_queue() was never
testing for this and passing a raw K_FOREVER down, where it got
interpreted as a negative timeout and caused it to fire at the next
tick.

Now that we actually see the original k_timeout_t here, we might as
well check it locally and do the correct thing (that is, nothing) if
asked to schedule a timeout that will never fire.

Fixes #244 [... cut]
closed 2020-04-22 by andrewboie