zephyrproject-rtos/zephyr #24414: Why there's no api for changing sensor's work mode?

zzzh, 2020-04-16:
**Is your enhancement proposal related to a problem? Please describe.**
yeah, I want to change adxl362's work mode, but found no api to do that.

**Describe the solution you'd like**
Add an api in sensor_driver_api array for changing sensor's work mode.

**Describe alternatives you've considered**
Not think of

**Additional context**
Many sensors have different work modes, so it's a common need to switch a sensor's work mode at run time.

carlescufi, 2020-04-21:
@zzzh for the time being you can also use driver-specific APIs. However, those will not currently work with userspace enabled. See https://github.com/zephyrproject-rtos/zephyr/issues/24539 for more info (cc @pabigot)

henrikbrixandersen, 2020-04-22:
Perhaps a solution could be to use `sensor_attr_set()` for setting the mode?

MaureenHelm, 2020-07-01:
Why do you want to change the work mode at runtime? This seems like something you would normally want to configure at build time.

closed 2020-09-18 by github-actions[bot]