zephyrproject-rtos/zephyr #24429: LPC55S69 flash faults when reading unwritten areas

d3zd3z, 2020-04-16:
Unlike nearly all Zephyr-supported devices, the internal flash memory on the LPC55S69 faults instead of returning a pattern (usually 0xFF or 0x00) on unwritten flash.

The drive implements read using `memcpy`, which on this device receives a fault instead of copying the data.

The flash controller is able to perform a read without faulting. It is undetermined whether the flash_read operation should return an error, or just read a pattern like most devices.

This is preventing MCUboot from working on this device, as MCUboot needs to be able to probe the flash to see if an image has been written there. MCUboot uses the flash driver, so making flash_read not fault, but either return an error, or read uninitialized data would be sufficient.

The easiest way to reproduce this is to build `samples/drivers/flash_shell` and read past the end of the programmed image. Most reliably, issue a full flash erase before programming (this depends on the flash tool, for the Segger, the `erase` command is sufficient).
```
uart:~$ flash read 0x40000 16
[00:00:21.801,000] <err> os: ***** BUS FAULT *****
[00:00:21.801,000] <err> os:   Precise data bus error
[00:00:21.801,000] <err> os:   BFAR Address: 0x40000
[00:00:21.801,000] <err> os: r0/a1:  0x30001060  r1/a2:  0x00040000  r2/a3:  0x00000010
[00:00:21.801,000] <err> os: r3/a4:  0x30001060 r12/ip:  0x30000171 r14/lr:  0x10008cc5
[00:00:21.801,000] <err> os:  xpsr:  0x81000200
[00:00:21.801,000] <err> os: Faulting instruction address (r15/pc): 0x100087b6
[00:00:21.801,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:21.801,000] <err> os: Current thread: 0x30000438 (unknown)
[00:00:21.870,000] <err> os: Halting system
```

(the code at 0x10008cc5 is the instruction following the call to `memcpy` within `flash_mcux_read`).

**Expected behavior**
The driver should either return a specific error, or read the uninitialized flash contents.

**Impact**
This prevents porting MCUboot to this target.

agansari, 2020-05-04:
@d3zd3z the short answer is it's an ECC error when reading an erased area.

LPC's flash firmware exposes additional checking functions: FLASH_VerifyErase and FLASH_VerifyProgram.
Checking FLASH_VerifyProgram the erased area returns a status of kStatus_FLASH_EccError, this is described in SoC's the user manual UM11126 - ch 5.7.13

> When reading an erased location, an uncorrectable error is flagged. Use the “blank
check” command to test for successful erase.

A possible solution would be use FLASH_VerifyErase before reading, in case the area was erased don't read it but do one of the following:
 - return EIO, ENOENT or some other error
 - write 0xFF or 0x00 to the data buffer

My comments would be the following:
 - an error that tells us the area is uninitialized should be used rather than one that tells us this is an unavailable area.
 - is writing blank data efficient? How does MCUboot check the are is blank?
 - optional verify erase and program functions in flash_driver_api would be useful, but I guess not all flash controllers hw have these functions, a SW implemented strcmp would hide this functionality.

d3zd3z, 2020-05-05:
> A possible solution would be use FLASH_VerifyErase before reading, in case the area was erased don't read it but do one of the following.

I don't think this is what we want. Remember, it is also valid for the flash to be in a partially erased or partially written state. I think read should just return an error in this case.

"Uninitialized" and "unavailable" are not very meaningful. Something like a flash filesystem will be tracking this itself. The only time it won't know what is in the area it last wrote to, and it should always try erasing that before writing to it again.

> is writing blank data efficient? How does MCUboot check the area is blank?

It doesn't. It looks for specific things, and if they are present uses them. MCUboot in particular works in two modes:

1. Normal upgrade check. It will look for a specific magic value, and then try to verify the signature of an image. Reading all 0xFF or all 0x00 would be fine in this case, as the signature would fail.
2. The upgrade process itself. This normally expects to be able to write to parts of the data that were erased by not written. Given this device writes to the entire page at a time, MCUboot will have to be modified to work with this flash. It can operate in an "overwrite" mode where it erases an entire region and then writes to it. As long as it has a way to read the data in flash without crashing, this mode would work with the device currently.

d3zd3z, 2020-05-05:
Kumar sent me a hacked version that just disables ECC when reading. This isn't really ideal, because we would like to operate with the ECC, and know if the data was unreadable. I think ideal would be for the read call to just do a regular read (through the controller), and if the data is unreadable, return an error from the read call instead of reading data.

agansari, 2020-05-11:
@d3zd3z I've ran low level commands on the flash memory controller, see below:
- CMD_BLANK_CHECK this is the one suggested by the manual, this would not be useful in partially erased area
- CMD_READ_SINGLE_WORD would only check 1 word at a time, there is not DMA like read command for more than a word, it's not safe unless we're looping for the whole area.
- CMD_MARGIN_CHECK I ended up using 

> If, for any reason, programming was interrupted or disturbed, or erase was performed without a subsequent programming, this check fails.

PR #25206 (2020-07-09): drivers: soc_flash_mcux check before reading
LPC55xxx SoCs don't allow reading erased areas, the flash memory
controller (FMC) will trigger an ECC error for that area. To prevent
reading unwritten areas the low level FMC command of margin checking is
used, this will fail if the area to be read is not correctly written.

Fixes #24429
closed 2020-07-09 by carlescufi