zephyrproject-rtos/zephyr #24437: smp_svr samle dosen't build for nrf52

nvlsianpu, 2020-04-17:
**Describe the bug**
Can't build smp_svr sample on nrf52 device

**To Reproduce**
Steps to reproduce the behavior:
1. ```west build -d build/smp_svr -b nrf52840dk_nrf52840 ./zephyr/samples/subsys/mgmt/mcumgr/smp_svr/```
2. build land in the error:
``` Console
/home/anpu/codedev/zephyrproject/modules/lib/mcumgr/cmd/os_mgmt/port/zephyr/src/zephyr_os_mgmt.c: In function 'os_mgmt_impl_reset':
/home/anpu/codedev/zephyrproject/modules/lib/mcumgr/cmd/os_mgmt/port/zephyr/src/zephyr_os_mgmt.c:94:66: error: incompatible type for argument 3 of 'k_timer_start'
   94 |     k_timer_start(&zephyr_os_mgmt_reset_timer, K_MSEC(delay_ms), 0);
      |                                                                  ^
      |                                                                  |
      |                                                                  int
In file included from /home/anpu/codedev/zephyrproject/zephyr/include/kernel.h:5570,
                 from /home/anpu/codedev/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/anpu/codedev/zephyrproject/modules/lib/mcumgr/cmd/os_mgmt/port/zephyr/src/zephyr_os_mgmt.c:20:
zephyr/include/generated/syscalls/kernel.h:389:92: note: expected 'k_timeout_t' {aka 'struct <anonymous>'} but argument is of type 'int'
  389 | static inline void k_timer_start(struct k_timer * timer, k_timeout_t duration, k_timeout_t period)
      |                                                                                ~~~~~~~~~~~~^~~~~~
[96/125] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
```

**Expected behavior**
sample builds and works

**Environment (please complete the following information):**
zephyr after https://github.com/zephyrproject-rtos/zephyr/pull/20983 PR was merged

nvlsianpu, 2020-04-17:
FYI @aunsbjerg @jukkar

nvlsianpu, 2020-04-17:
cc @de-nordic

nvlsianpu, 2020-04-17:
`CONFIG_LEGACY_TIMEOUT_API=y` fixes the issue.
It should be selected unless https://github.com/zephyrproject-rtos/zephyr/pull/24341 not merged.

aunsbjerg, 2020-04-17:
@nvlsianpu Is there a case for building the SMP sample without any of the overlay configs? The base config does not enable any of the SMP transports, which renders the sample pretty useless as far as I can tell.

This works: 
`west build -d build/smp_svr -b nrf52840dk_nrf52840 ./zephyr/samples/subsys/mgmt/mcumgr/smp_svr/ -- -DOVERLAY_CONFIG=overlay-bt.conf`

Running sanitycheck on the samples also works: 
`./zephyr/scripts/sanitycheck -T zephyr/samples/subsys/mgmt/mcumgr/smp_svr/ -c` builds fine.

commit 5c8a2eba (2020-04-17): west.yml: Snapshot update of mcumgr from upstream

This commit updates mcumgr with latest snapshot from upstream.

Fixes #24437
closed 2020-04-17 by ioannisg