zephyrproject-rtos/zephyr #24442: samples/subsys/mgmt/mcumgr/smp_svr: should enable BT and FS for nrf52 boards

nvlsianpu, 2020-04-17:
**Describe the bug**
The sample used to build with BT transport and FS command set enabled for out-of the box build. After merge https://github.com/zephyrproject-rtos/zephyr/pull/20983 its stops.

**To Reproduce**
Steps to reproduce the behavior:
1.` west build -d build/smp_svr -b nrf52840dk_nrf52840 ./zephyr/samples/subsys/mgmt/mcumgr/smp_svr/`

**Expected behavior**
These feature should be enabled by default for nrf52 targets

**Impact**
bad user experience

**Environment (please complete the following information):**
 - runing master https://github.com/zephyrproject-rtos/zephyr/commit/762b5219310e5e8acca0d02bab4c3f883f5a4d81

aunsbjerg, 2020-04-17:
The sample was intentionally split up so that prj.conf contained only the common configuration set and transport specific configuration is contained in overlay configurations. To enable bluetooth transport, you'll need to add one of the bt overlay configs in the build command, like so: 
`west build -d build/smp_svr -b nrf52840dk_nrf52840 ./zephyr/samples/subsys/mgmt/mcumgr/smp_svr/ -- -DOVERLAY_CONFIG=overlay-bt.conf`

As for the missing FS command set, that's a bug and I will make a PR to fix that.

nvlsianpu, 2020-04-17:
I supposed that this sample is building with required features enables out of the box for these boards.
From my experience can avoid most of "help-desk" question thanks to that.

PR #24448 (2020-04-20): samples/susbys/mgmt/mcumgr/smp_svr: add nrf5 configs
Added default configuration files for nrf5 boards so the sample
is building with expected features for these boards.

Would be greater to use OVERLAY_CONFIG instead of files addition,
but that doesn't work from CMakeLists.txt scope.

fixes #24442
commit 8579d9f3 (2020-04-20): samples/susbys/mgmt/mcumgr/smp_svr: add nrf5 configs

Added default configuration files for nrf5 boards so the sample
is building with expected features for these boards.

Would be greater to use OVERLAY_CONFIG instead of files addition,
but that doesn't work from CMakeLists.txt scope.

fixes #24442
closed 2020-04-20 by carlescufi