zephyrproject-rtos/zephyr #24453: docs: Allow to use the exact current zephyr version number in place of /latest/ in documentation URLs

KubaFYI, 2020-04-17:
**Is your enhancement proposal related to a problem? Please describe.**

Currently when `/latest/` in Zephyr docs URL (`https://docs.zephyrproject.org/latest/*`) is substituted with `<MAJOR>.<MINOR>.<PATCH>` of the most recent release (like one would reach docs for older releases) a `Page Not Found` page is being shown. This creates a problem in a situation when the most recent release changes, and links used externally in mind with the old release version start pointing to that new release.

This is problematic in situations like writing set-up instructions for an external project using a specific version of Zephyr, which at the time of writing is the latest release.

**Describe the solution you'd like**
Have `/latest/` in docs URL be interchangeable with the actual version number of that release, in the same way one can use older release version numbers.

carlescufi, 2020-04-21:
@KubaFYI the problem is that if we do that you'd get the false impression that the doc corresponds to the release, and it really doesn't: it instead just mirrors the current state of master at the time the docs were published (docs are built and published once a day).
One thing we could do is to use the .99 notation and have `2.3.99` redirect to `latest`.

CC @nashif

nashif, 2020-04-21:
I do not follow what the issue is...

stephanosio, 2020-04-22:
The latest Zephyr release, at the time of writing, is 2.2.0 and https://docs.zephyrproject.org/2.2.0/ is working just fine.

KubaFYI, 2020-04-22:
@carlescufi Thanks for explaining this, it seems I misunderstood what the website claimed - I assumed that the 'latest' which right now corresponds to 2.2.99 was an actual release (I should have checked that). To be fair in the docs menu 'latest' figures under a "Document Release Versions" header, so this might give people the wrong idea that the current state is a release and not unset and changing master it actually is.

[image]

nashif, 2020-04-22:
I think what we could fix is remove the `2.2.99` and put `latest` instead in the picture above.

carlescufi, 2020-05-06:
@nashif is this something we can do or does it require changes in the web server or sphinx?

nashif, 2020-08-14:
it is something that can be done in the doc source code, templates...

PR #28678 (2020-10-19): doc: use latest label instead of version in docs
Instead of having a .99 version, use latest and link to the lates
documentation on the left side bar.

The version number has been confusing users where it was difficult to
determine if they were looking at the development tree or some released
version.

Fixes #24453
commit 5d0cf010 (2020-10-19): doc: use latest label instead of version in docs

Instead of having a .99 version, use latest and link to the lates
documentation on the left side bar.

The version number has been confusing users where it was difficult to
determine if they were looking at the development tree or some released
version.

Fixes #24453
closed 2020-10-19 by nashif