zephyrproject-rtos/zephyr #24484: The file system shell example failed to build

jasonedn, 2020-04-18:
**Describe the bug**
Failed to build the file system shell example on native_posix,

**To Reproduce**
Steps to reproduce the behavior:
1. export PKG_CONFIG_PATH=/usr/lib/i386-linux-gnu/pkgconfig
2. west build -b native_posix samples/subsys/shell/fs

**Expected behavior**
Pass the building

**Impact**

**Screenshots or console output**
...
ephyr && cmake -E rename zephyr_prebuilt.map zephyr.map && /usr/bin/objdump -S zephyr.elf > zephyr.lst && /usr/bin/readelf -e zephyr.elf > zephyr.stat && /usr/bin/cmake -E copy zephyr.elf zephyr.exe
/usr/bin/ld: skipping incompatible //usr/lib/x86_64-linux-gnu/libfuse.so when searching for -lfuse
/usr/bin/ld: skipping incompatible //usr/lib/x86_64-linux-gnu/libfuse.a when searching for -lfuse
/usr/bin/ld: cannot find -lfuse
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build 
...

**Environment (please complete the following information):**
 - OS: 64-bit Ubuntu 18.04
 - Toolchain:  Zephyr SDK
 - Commit SHA : 2b29d3

**Additional context**
This issue is related to libfuse.  The libfuse-dev was installed from https://ubuntu.pkgs.org/18.04/ubuntu-main-i386/libfuse-dev_2.9.7-1ubuntu1_i386.deb.html.
However the folder "/usr/lib/i386-linux-gnu/pkgconfig" couldn't be found anywhere.

pabigot, 2020-04-18:
Seems likely the README is out of date.

If you do:

    west build -b native_posix_64 samples/subsys/shell/fs/

does it work?

jasonedn, 2020-04-19:
> Seems likely the README is out of date.
> 
> If you do:
> 
> ```
> west build -b native_posix_64 samples/subsys/shell/fs/
> ```
> 
> does it work?

No, it doesn't work. According to the online document, it seems that native-posix supports 32-bit only:
"Before starting a build, make sure that the i386 pkgconfig directory is in your search path and that a 32-bit version of libfuse is installed. For more background information on this requirement see Native POSIX execution (native_posix)."

I spent hours to try to install 32-bit libfuse library, but it failed. Later I found 64-bit support could be configured in menuconfig as below:
[image]

closed 2020-04-19 by jasonedn

pabigot, 2020-04-19:
Yes, `native_posix` is 32-bit, but `native_posix_64` is 64-bit.  Which should have worked immediately.

overheat, 2020-04-21:
@pabigot  I have tried this:
```
sudo apt install libfuse-dev
west build -b native_posix_64 samples/subsys/shell/fs/
west build -t run
```
I can see this,
```
Mounting flash at flash/
UART_0 connected to pseudotty: /dev/pts/2
*** Booting Zephyr OS build zephyr-v2.2.0-1666-gf3b4d8a86f78  ***
```
but after `screen /dev/pts/2`, no shell, nothing display on the screen.

pabigot, 2020-04-21:
@overheat I can confirm that behavior, but have no idea what the problem is.  Please open a question issue.

overheat, 2020-04-21:
Sure.
Thanks!

Best regards,
Aaron