zephyrproject-rtos/zephyr #24485: k_pipe_get() does not respect min_xfer

cfriedt, 2020-04-19:
**Describe the bug**
The following code hangs indefinitely at `k_pipe_get()`, but it should not, according to [the documentation](https://docs.zephyrproject.org/latest/reference/kernel/data_passing/pipes.html):
```
unsigned char buf[64];
struct k_pipe;
size_t bytes_written = 0;
size_t bytes_read = 0;
char rxbuf[32] = {};

k_pipe_init(&p, buf, sizeof(buf));

k_pipe_put(&p, "Hi!", 3, &bytes_written, 3 /* min_xfer */, K_FOREVER);
// wrote 3 bytes
printf("Wrote Hi!\n");

k_pipe_get(&p, rxbuf, sizeof(rxbuf), &bytes_read, 1 /* min_xfer */, K_FOREVER);
// bytes_read should equal 3
// k_pipe_get() should return immediately since min_xfer is 1
// but hangs indefinitely
printf("Read '%s'\n", rxbuf);
```
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=qemu_cortex_m3
3. make
4. See error

**Expected behavior**

As per [the documentation](https://docs.zephyrproject.org/latest/reference/kernel/data_passing/pipes.html)

> Data can be synchronously received from a pipe by a thread. If the specified minimum number of bytes can not be immediately satisfied, then the operation will either fail immediately or attempt to receive as many bytes as possible and then pend in the hope that the receive can be completed later. Accepted data is either copied from the pipe’s ring buffer or directly from the waiting sender(s).

Since "the specified minimum number of bytes can be immediately satisfied", the call to `k_pipe_get()` should return immediately.

**Impact**
This is a showstopper for me if I would like to continue using the k_pipe API.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: cross-compile
 - Commit SHA or Version used: 2833d016aa0b747c5c6fdd363dea4a9ebda941eb

**Additional context**

I can make a PR. Have a patch for it already. I wonder though if `k_pipe_put()` has the same problem. Will investigate.

cfriedt, 2020-04-19:
Yep. `k_pipe_write()` has the same problem. Will include a fix for that as well.

pabigot, 2020-04-19:
While I agree the documentation could be interpreted the way you do, and that it might be useful if it worked that way, it seems the implementation respects "immediately satisfied" and "fail immediately" only in the case of `K_NO_WAIT`.

If your interpretation is to be supported I don't think your proposed solution in #24486 will work because at the point you discover that the minimum transfer hasn't been satisfied immediately some bytes may have been transferred, and they can't be clawed back, so failure will cause lost data.

I suspect a proper solution would be to modify `pipe_xfer_prepare()`, which checks for how much is available immediately but also has comments that are inconsistent with your desired interpretation.  Additional  modifications in the other functions will be required to complete early without pending when `min_xfer` is satisfied.

But first it has to be confirmed that this is a bug, rather than misleading documentation.

cfriedt, 2020-04-19:
I guess technically, if K_NO_WAIT is specified, then the documentation is consistent.

However, I see no way to perform blocking I/O without entirely pre-determined data lengths. To be completely transparent, it is #24426 that is blocked.

cfriedt, 2020-04-20:
I interpreted the API documentation to mean that if I called `k_pipe_get()` with `min_xfer = 1` and `timeout = K_FOREVER`, then I would be able to mimic the same behaviour as read(2), and perform an actual blocking call that would return as soon as min_xfer was received or an error occurred.

Technically speaking, if an error occurs and some bytes have been read, the the number of bytes read is still available in `bytes_read`, so they don't need to necessarily be "clawed back".

carlescufi, 2020-04-21:
@andyross could you take a look?

andyross, 2020-04-21:
Yeah, that "min_xfer" is an ugly wart.  But I think I agree with @cfriedt that it can only reasonably be interpreted as having higher precedence than the timeout.  That is, what we **want** is:

"Here is some data (or a buffer to receive data), please transfer at least this many bytes of it, if necessary waiting this long to do it before returning"

But what we've implemented is:

"Here is some data (or buffer), please transfer all of it, if necessary waiting this long before returning, **but** as a special case it's OK to return early if you can synchronously move this smaller amount instead".

Yeah, we're wrong, even if you can squint and make us legal per our vague docs.  The PR looks good, I'll put comments there.

cfriedt, 2020-04-21:
Thanks for having a look @andyross @carlescufi

PR #24486 (2020-04-28): kernel: pipe: if timeout is not K_NO_WAIT and >= min_xfer bytes transferred, then return immediately
If `timeout != K_NO_WAIT`, then return immediately when not all `bytes_to_read` / `bytes_to_write` have been transfered, but `>= min_xfer` have been transferred.

Fixes #24485
commit d650f4b4 (2020-04-28): kernel: pipe: fix !K_NO_WAIT and >= min_xfer bytes transferred

If timeout != K_NO_WAIT, then return immediately when not all
bytes_to_read or bytes_to_write have been transfered, but >=
min_xfer have been transferred.

Fixes #24485
closed 2020-04-28 by carlescufi