zephyrproject-rtos/zephyr #24490: SPI-NOR driver not found in spi_flash sample

akofoed, 2020-04-19:
**Describe the bug**
Testing the SPI-NOR driver (jedec,spi-nor), my code says it cannot find the driver. So I tried the sample in samples/drivers/spi_flash/ and it too cannot find the driver. I have tested this on both current master and v2.2.0.

**To Reproduce**
To see the error you don't need the spi-nor flash attached to your board. It is the spi-nor driver that it cannot find.
I am using the nucleo_429zi board and this is my addition to the DTS:
```
&spi1 {
	status = "okay";
	clock-frequency = <10000000>;
	cs-gpios = <&gpioe 11 0>;
	flash1: mx25r6435f@0 {
		compatible = "jedec,spi-nor";
		status = "okay";
		label = "MX25R64";
		spi-max-frequency = <10000000>;
		reg = <0>;
		jedec-id = [c2 28 17];
		size = <67108864>;
		wp-gpios = <&gpioe 9 0>;
		hold-gpios = <&gpioe 10 0>;
	};
};

&flash1 {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		nvs_partition: partition@0 {
			label = "nvs-partition";
			reg = <0x00000000 0x00010000>;
		};
	};
};
```
Yes, I am using spi4 (gpioE11->E14) but I get the same error as in the sample that the driver cannot be found.
This is the output using the current master (just updated):
```
*** Booting Zephyr OS build zephyr-v2.2.0-1677-ge7c7f911a9cd  ***

JEDEC SPI-NOR SPI flash testing
==========================
SPI flash driver MX25R64 was not found!
```
Output on v2.2.0:
```
*** Booting Zephyr OS build zephyr-v2.2.0  ***

JEDEC SPI-NOR SPI flash testing
==========================
SPI flash driver MX25R64 was not found!
```

**Expected behavior**
I would have expected the driver to be found.

**Impact**
External SPI-flash not working.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
 - OS: Linux/Ubuntu
 - Toolchain Zephyr SDK
 - Commit Version: Current master and v2.2.0

**Additional context**
None.

pabigot, 2020-04-21:
I don't have NXP hardware with a similar flash to test, but the "not found" will be received if driver initialization failed.  This can be because:
* the associated SPI bus was not found (your overlay shows it's enabled in devicetree, so this should be ok);
* The chip-select is on a GPIO (rather than pinmux) and its device is not correct;
* An attempt to read the JEDEC ID failed;
* You've enabled `CONFIG_SPI_NOR_IDLE_IN_DPD` and dpd failed (not likely).

I suspect the problem is due to not reading the JEDEC ID; you can confirm this by adding `printk` statements to the `drivers/flash/spi_nor.c` function `spi_nor_configure()`.

If you have `CONFIG_SPI_STM32_USE_HW_SS=y`, which appears to be the default, then SPI will be attempting to use a hardware-specified chipselect provided by pinmux, not whatever you put into `cs-gpios`.  The pinmux file for this board indicates that PA4 is the required chipselect line.  Is that what you've wired to your flash device?

akofoed, 2020-04-22:
@pabigot 
Thank you for your pointers, you were right in that the attempt to read the JEDEC ID failed. I capitulated and had to pull out my Saelae logic probe and it turned out that I had a loose connection and had to play with some settings regarding SPI-CS to get it working. It is working now except that sector size is wrong:
`rc = flash_get_page_info_by_offs(nvs_device, db_fs.offset, &info);`
gives me a info.size of 65536 (decimal) which should be 4KByte.
I will close this as this is not a bug.
Sorry about this.
Thanks

closed 2020-04-22 by akofoed