zephyrproject-rtos/zephyr #24495: devicetree: add DT_HAS_ALIAS()

sslupsky, 2020-04-19:
It would be useful to have a DT_HAS_ALIAS() macro in devicetree.h similar to DT_HAS_NODE() for the situation where you use an alias to select the item from the device tree.

My example use case is debug pins where the pins are not dedicated for debug but are transient.  So, the debug pins overlap existing GPIOs in the devicetree.  I would like to use an alias for the debug pins and alias them to the existing GPIO pins.  When I need to use them, I enable the alias in the devicetree overlay.  When I no longer want to use them, I remove them from the overlay.

sslupsky, 2020-04-19:
I think this will work
```
#define DT_HAS_ALIAS(alias_name) DT_HAS_NODE(DT_ALIAS(alias_name))
```

henrikbrixandersen, 2020-04-19:
Why not simply use `DT_HAS_NODE(DT_ALIAS(alias_name))`?

mbolivar-nordic, 2020-04-20:
DT_HAS_NODE(DT_ALIAS(foo)) is a valid implementation, yes. DT_ALIAS(foo) is a node identifier like any other, and DT_HAS_NODE() works with any node identifier.

@sslupsky would you be willing to send a patch? You could add this under DT_HAS_CHOSEN() in devicetree.h. Just make sure to add test cases in tests/lib/devicetree/src/main.c.

sslupsky, 2020-04-20:
Sure, I can do that.

kartben, 2023-08-21:
> Sure, I can do that.

Are you still interested? :)

rruuaanng, 2024-11-17:
It has merged :). I'll close this issue

rruuaanng, 2024-11-17:
https://github.com/zephyrproject-rtos/zephyr/pull/80511

closed 2024-11-17 by rruuaanng