zephyrproject-rtos/zephyr #24497: frdm_k64f fatal error while using flash and TLS features together

timdesi, 2020-04-19:
**Describe the bug**
I am developing full-stack application that connects to cloud with MQTT TLS. Need to store same basic variables like username, password to flash/nvs directly or with settings.

Problem is that using flash/nvs with TLS enabled configuration breaks flash memory and causes fatal os error. 

What have you tried to diagnose or workaround this issue?

All tested samples like subsys/settings, subsys/nsv, subsys/fs/littlefs, net/cloud/*, net/sockets/* stand 
alone run as expected.

But when you add to sample conf. file CONFIG_NET_SOCKETS_SOCKOPT_TLS=y option boards fails.

Qemu works us expected.

**To Reproduce**
Steps to reproduce the behavior:

For to be simple, i have choose subsys/fs/littlefs sample.

1. mkdir build; cd build
2. cmake -GNinja -DBOARD="frdm_k64f" ..
3. Add same TLS conf to prj.conf file

```
CONFIG_MPU_ALLOW_FLASH_WRITE=y

# Enable networking
CONFIG_NETWORKING=y

# Disable IPv6 support
CONFIG_NET_IPV6=n

# Enable IPv4 support
CONFIG_NET_IPV4=y
CONFIG_NET_IF_MAX_IPV4_COUNT=2
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=3

# Enable TCP support
CONFIG_NET_TCP=y

# Enable Sockets support
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_SOCKOPT_TLS=y
CONFIG_NET_SOCKETS_TLS_MAX_CREDENTIALS=1

# Enable MQTT Lib support
CONFIG_MQTT_LIB=y
CONFIG_MQTT_LIB_TLS=y

# Enable Mbed TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=n
CONFIG_MBEDTLS_HEAP_SIZE=100000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=10240
CONFIG_MBEDTLS_PEM_CERTIFICATE_FORMAT=y

CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
```
4. ninja flash
4. See error

**Expected behavior**
Store basic settings to flash and connect to cloud via TLS.

**Impact**
Could not continue development with frdm_k64f board.

**Screenshots or console output**

After first flash board, fs mounts with same errs but after reset starts to gove fatal error.

```
*** Booting Zephyr OS build zephyr-v2.2.0-1526-g40ee38a94505  ***
Cert reg OKArea 1 at 0x1e000 on FLASH_CTRL for 8192 bytes
Erasing flash area ... 0
/lfs mount: 0
/lfs: bsize = 16 ; frsize = 4096 ; blocks = 2 ; bfree = 0
/lfs/boot_count stat: -2
/lfs/boot_count read count 0: 0
/lfs/boot_count seek start: 0
/lfs/boot_count write new boot count 1: 4
/lfs/boot_count close: 0
/lfs opendir: 0
  F 4 boot_count
End of files
/lfs unmount: 0
[00:00:00.028,000] <inf> littlefs: LittleFS version 2.1, disk version 2.0
[00:00:00.028,000] <inf> littlefs: FS at FLASH_CTRL:0x1e000 is 2 0x1000-byte blocks with 512 cycle
[00:00:00.028,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:00:00.028,000] <err> littlefs: Corrupted dir pair at 0 1
[00:00:00.028,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:00:00.038,000] <inf> littlefs: /lfs mounted
[00:00:00.045,000] <err> fs: failed get file or dir stat (-2)
[00:00:00.063,000] <inf> littlefs: /lfs unmounted
*** Booting Zephyr OS build zephyr-v2.2.0-1526-g40ee38a94505  ***
[00:00:00.005,000] <err> os: ***** USAGE FAULT *****
[00:00:00.005,000] <err> os:   Attempt to execute undefined instruction
[00:00:00.005,000] <err> os: r0/a1:  0x200011a4  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:00.005,000] <err> os: r3/a4:  0x20000dcc r12/ip:  0x00000001 r14/lr:  0x00001d41
[00:00:00.005,000] <err> os:  xpsr:  0x61000000
[00:00:00.005,000] <err> os: Faulting instruction address (r15/pc): 0x0001eff2
[00:00:00.005,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.005,000] <err> os: Current thread: 0x20002354 (unknown)
[00:00:00.063,000] <err> os: Halting system
*** Booting Zephyr OS build zephyr-v2.2.0-1526-g40ee38a94505  ***
[00:00:00.005,000] <err> os: ***** USAGE FAULT *****
[00:00:00.005,000] <err> os:   Attempt to execute undefined instruction
[00:00:00.005,000] <err> os: r0/a1:  0x200011a4  r1/a2:  0x00000000  r2/a3:  0x00000000
[00:00:00.005,000] <err> os: r3/a4:  0x20000dcc r12/ip:  0x00000001 r14/lr:  0x00001d41
[00:00:00.005,000] <err> os:  xpsr:  0x61000000
[00:00:00.005,000] <err> os: Faulting instruction address (r15/pc): 0x0001eff2
[00:00:00.005,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.005,000] <err> os: Current thread: 0x20002354 (unknown)
[00:00:00.063,000] <err> os: Halting system

```

**Environment (please complete the following information):**
 - OS: CentOS 7
 - Toolchain : Zephyr SDK
 - Commit SHA or Version used : zephyr-v2.2.0-1526-g40ee38a94505

marketideas, 2020-04-20:
Hi, I have found the .dts for the FRDM-K64F creates a storage partition somewhere in the middle of the flash, and not at the end as expected.  Once your code grows and wraps into that partition, the NV_SETTINGS stuff will erase code, and this is a bad thing. TLS is quite large. I can't guarantee this is your issue, but it certainly could be and get you moving forward again.  Are you bringing in MBEDTLS as well?  If so, you might encounter some heap issues, which I have yet to investigate and solve. (MBEDTLS (with Openthread) seems to want to use a fake heap that is only 8 bytes long.

timdesi, 2020-04-21:
Moving storage partition at the end of flash solves this problem.

closed 2020-04-21 by timdesi