zephyrproject-rtos/zephyr #24499: devicetree:  node name for SPI buses should be 'spi' warning

sslupsky, 2020-04-19:
I have noticed a warning when I build my project:
```
arduino_mkr_1310.dts.pre.tmp:110.28-115.5: Warning (spi_bus_bridge): /soc/sercom@42000c00: node name for SPI buses should be 'spi'
  also defined at arduino_mkr_1310.dts.pre.tmp:211.10-213.3
  also defined at arduino_mkr_1310.dts.pre.tmp:264.10-271.3
  also defined at arduino_mkr_1310.dts.pre.tmp:369.10-388.3
arduino_mkr_1310.dts.pre.tmp:116.28-121.5: Warning (spi_bus_bridge): /soc/sercom@42001000: node name for SPI buses should be 'spi'
  also defined at arduino_mkr_1310.dts.pre.tmp:214.10-216.3
  also defined at arduino_mkr_1310.dts.pre.tmp:272.10-292.3
<stdout>: Warning (spi_bus_reg): Failed prerequisite 'spi_bus_bridge'
```
Here are the lines 110 - 121 from the pre.tmp file:
```
  sercom1: sercom@42000c00 {
   compatible = "atmel,sam0-sercom";
   reg = <0x42000c00 0x40>;
   status = "disabled";
   label = "SERCOM1";
  };
  sercom2: sercom@42001000 {
   compatible = "atmel,sam0-sercom";
   reg = <0x42001000 0x40>;
   status = "disabled";
   label = "SERCOM2";
  };
```
I am not sure if this can be ignored or if there is something that needs to be fixed here?  It appears to complain about the "sercomx" node id.  The sam0 sercom modules can be configured as a UART, SPI or I2C.

Does anyone have a suggestion about how to fix the warning?

sslupsky, 2020-06-30:
@carlescufi @galak Any thoughts on how to fix the warning?

carlescufi, 2020-07-10:
> @carlescufi @galak Any thoughts on how to fix the warning?

No, but I have added @mbolivar-nordic in the hope that he can give you feedback on it.

mbolivar-nordic, 2020-07-10:
You can add dtc flags on a board-specific basis using pre_dt_board.cmake. See boards/arm/lpcxpresso54114/pre_dt_board.cmake for example. My guess is that adding `-Wno-spi_bus_bridge` to the dtc flags will silence the warning.

Perhaps we may need a way to do this on an SoC-wide basis, but I would like @galak to confirm that'd be a good idea in this case. It seems reasonable to me that the "Generic Names Recommendation" DT spec (which is where this warning looks like it's coming from) is safe to silence in this case.

sslupsky, 2020-07-18:
@mbolivar Thank you for the suggestion, adding the flag cleared up the warning.

I will leave this issue open until @galak has a chance to comment.

decsny, 2023-08-21:
re-opening because just encountered, the same issue with flexcomms on nxp mcus, and I am wondering 1) if there is a way to make this SOC-wide, and 2) is it possible to optionally escalate dtc warnings into errors

mbolivar-ampere, 2023-08-28:
> 1) if there is a way to make this SOC-wide, 

Not today.

> and 2) is it possible to optionally escalate dtc warnings into errors

Yes, but be careful here: `dtc` is an optional dependency and the build system won't use it if it's not installed, so in general it cannot be relied upon to detect errors.

You can define the cmake variable `EXTRA_DTC_FLAGS` to a cmake `;-list` of flags you want to pass to dtc. For example, `-DEXTRA_DTC_FLAGS='-Eunit_address_vs_reg;-Eaddress_cells_is_cell'`. The dtc error flags are not documented. You need to read the dtc source code in checks.c for details on how they work.

You can define the cmake variable `EXTRA_GEN_DEFINES_ARGS` to `--edtlib-Werror` to turn all edtlib warnings into errors. This can be relied upon to halt the build in all situations when our dt tooling causes a warning, but the set of warnings is more limited. Extending that set is outside the scope of this issue in my opinion.