zephyrproject-rtos/zephyr #24500: Failed to run the sample "Native Posix Ethernet"

jasonedn, 2020-04-19:
**Describe the bug**
The sample "Native Posix Ethernet" failed to run after build

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b native_posix samples/net/eth_native_posix
2. sudo --preserve-env=ZEPHYR_BASE make -Cbuild run  
  or
  sudo --preserve-env make -Cbuild run

**Expected behavior**
It runs successfully

**Impact**

**Screenshots or console output**
...
make: Entering directory '/home/jason/zephyrproject/zephyr/build'
make: *** No rule to make target 'run'.  Stop.
make: Leaving directory '/home/jason/zephyrproject/zephyr/build'
...

**Environment (please complete the following information):**
 - OS:  Ubuntu 18.04 LTS
 - Toolchain Zephyr SDK
 - Commit SHA: a42617

**Additional context**

overheat, 2020-04-19:
```
Expected behavior
It runs successfully
```

what is the issue?

overheat, 2020-04-19:
1.west build -b native_posix samples/net/eth_native_posix
2.west build -t run/ ./build/zephyr/zephyr.exe 
works for me.

Maybe this is an issue for the second step?

carlescufi, 2020-04-20:
@jasonedn as @overheat said, you are not running the right commands. Please try those and let us know.

closed 2020-04-20 by carlescufi

jasonedn, 2020-04-20:
@Aaron and@Carles . It still failed at the second step as below:

 $ west build -p auto -b native_posix samples/net/eth_native_posix
-- west build: generating a build system
...
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jason/zephyrproject/zephyr/build
-- west build: building application
[1/138] Preparing syscall dependency handling

[138/138] Linking C executable zephyr/zephyr.elf

$ west build -t run/ ./build/zephyr/zephyr.exe
  ERROR: source directory ./build/zephyr/zephyr.exe does not exist
  FATAL ERROR: refusing to proceed without --force due to above error

Actually "zephyr.exe" exists there and it started properly with the
following:

$ ./build/zephyr/zephyr.exe
UART_0 connected to pseudotty: /dev/pts/1
WARNING: Using a test - not safe - entropy source
*** Booting Zephyr OS build zephyr-v2.2.0-1690-g394a4443004b  ***

It didn't worked as the instruction in the online documents, and they
should be updated.

overheat, 2020-04-21:
@jasonedn `west build -t run` and `./build/zephyr/zephyr.exe` both  works.
After that you need to _UART_0 connected to pseudotty: /dev/pts/1_, like `screen /dev/pts/1`