zephyrproject-rtos/zephyr #24505: Configuration messages: Config SIG Model Subscription Get, Config SIG Model Subscription List

robinh17, 2020-04-20:
**Problem**
Nodes implemented with the Configuration Client model needs to be able to send a "Config SIG Model Subscription Get" message or "Config Vendor Model Subscription Get". On the target node, the configuration server model shall reply with a "Config SIG Model Subscription List" or "Config Vendor Model Subscription List" containing all addresses from the Subscription List of the model. The Bluetooth mesh APIs does not have a function for this. 

**Possible solution**
Bluetooth mesh configuration client model:
-Add function that gets the Subscription List from target node. 
-Add model opcode handler for ("Config SIG Model Subscription List", 0x80 0x2A)
-Add model opcode handler for ("Config Vendor Model Subscription List", 0x80 0x2C)

Bluetooth mesh configuration server model:
-Add model opcode handler for("Config SIG Model Subscription Get, 0x80 0x29)
-Add model opcode handler for("Config Vendor Model Subscription Get, 0x80 0x2B)

jhedberg, 2020-04-20:
The server side of this is already implemented, as far as I can see

LingaoMeng, 2020-04-21:
it's true, the cfg_srv side already implemented. see https://github.com/zephyrproject-rtos/zephyr/blob/c97f6f8eeb06c20c75b2c9a7690d9c58f3ea07af/subsys/bluetooth/mesh/cfg_srv.c#L3175-L3181

trond-snekvik, 2020-04-22:
All the "get list" type config client messages are missing.

I can take a look at this.

closed 2020-04-23 by robinh17

robinh17, 2020-04-23:
Fixed in #24594

PR #24594 (2020-04-23): Bluetooth: Mesh: Add cfg cli list getters
Adds config client getters for all list commands:
- mod_app_get
- mod_sub_get
- net_key_get
- app_key_get

Closes #24505.
commit 86137c3a (2020-04-23): Bluetooth: Mesh: Add cfg cli list getters

Adds config client getters for all list commands:
- mod_app_get
- mod_sub_get
- net_key_get
- app_key_get

Closes #24505.