zephyrproject-rtos/zephyr #24509: Ethernet Sample Echo Failed in Nucleo_f429zi - bisected

giellamoswhard, 2020-04-20:
**Describe the bug**
I've tried with commit 65eacef644 to use the IP connection on a nucleo_f429zi but it doesn't work.
I bisected it and it seems the problem slipped in with commit 
`commit bfaa1091fd6bc1877992bd40973adfea1412603b
Author: Erwan Gouriou <erwan.gouriou@linaro.org>
Date:   Thu Apr 16 20:52:22 2020 +0200

    boards: stm32: Use dt API for peripheral configuration

    Replace use of Kconfig SPI_X symbols by calls to DT API.

    Signed-off-by: Erwan Gouriou <erwan.gouriou@linaro.org>`

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyrproject/zephyr/samples/net/sockets/echo
2. west build -p auto -b nucleo_f429zi .
3. west flash
4. from pc ping 192.0.2.1

**Expected behavior**
Example must work

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

ABOSTM, 2020-04-21:
Hi @giellamoswhard,
With the commit you highlight, there is a conflict for STM32F429 nucleo board:
Both SPI1 and Ethernet use pin PA7.
As a workaround, you can disable SPI1: edit file ...\boards\arm\nucleo_f429zi\nucleo_f429zi.dts
```
&spi1 {
	status = "disabled";
};
```
I need to wait @erwango , to define the best solution to solve this conflict.

giellamoswhard, 2020-04-21:
@ABOSTM Thank you very much the workaround is confirmed to work.

akofoed, 2020-04-21:
You can use MOSI on pin PB5 instead:
pinmux:
```
#ifdef CONFIG_SPI_1
#ifdef CONFIG_SPI_STM32_USE_HW_SS
	{STM32_PIN_PA4, STM32F4_PINMUX_FUNC_PA4_SPI1_NSS},
#endif /* CONFIG_SPI_STM32_USE_HW_SS */
	{STM32_PIN_PA5, STM32F4_PINMUX_FUNC_PA5_SPI1_SCK},
	{STM32_PIN_PA6, STM32F4_PINMUX_FUNC_PA6_SPI1_MISO},
	{STM32_PIN_PB5, STM32F4_PINMUX_FUNC_PB5_SPI1_MOSI},
#endif	/* CONFIG_SPI_1 */
```

akofoed, 2020-04-21:
I see that the last part of the .../nucleo_f429zi/pinmux.c that someone has updated all the ifdefs to DT_HAS_NODE but they are missing the last USB_otg_fs one. This compiles on my machine:

```
#if DT_HAS_NODE(DT_NODELABEL(usbotg_fs))
	{STM32_PIN_PA11, STM32F4_PINMUX_FUNC_PA11_OTG_FS_DM},
	{STM32_PIN_PA12, STM32F4_PINMUX_FUNC_PA12_OTG_FS_DP},
#endif	/* CONFIG_USB_DC_STM32 */
```

PR #25288 (2020-05-20): boards: stm32: Restore Kconfig control on pinmux files
Move to new DT API removed the Kconfig control on individual peripheral activation which was used to control board pin configuration and allowed to avoid conflicting usages of pin between peripherals.

While it is good to use dts as a reference for peripheral instances enabling in the sw, a Kconfig notion should be controlled in files that are compiled whatever the use case. 

For all stm32 based boards, add Kconfig subystem control on pin configuration:
 `#if DT_NODE_HAS_STATUS(DT_NODELABEL(spi2), okay) && CONFIG_SPI`

Fixes #24509 (/#252 [... cut]
closed 2020-05-20 by carlescufi