zephyrproject-rtos/zephyr #24512: set OVERLAY_CONFIG as cmake variable

nvlsianpu, 2020-04-20:
**Is your enhancement proposal related to a problem? Please describe.**
 OVERLAY_CONFIG file is something used currently for creating children configuration basing on general parent configuration.
currently OVERLAY_CONFIG is possible to be used form command-line and from tests/sampes *.yaml description.

I.m asking for extend this to be used by CMakeLists explicitly as well.
This helps to reduce amount of configuration per boards we have in the codebase and might help to share same configuration between similar platforms.

```cmake
if((BOARD STREQUAL nrf52840dk_nrf52840) OR
   (BOARD STREQUAL nrf52dk_nrf52832))
  set(OVERLAY_CONFIG overlay-bt.conf)
endif()
```

aunsbjerg, 2020-04-21:
This is already supported, but apparently it requires the OVERLAY_CONFIG variable to be set before the zephyr boilerplate is included, like so:

```
if (${BOARD} STREQUAL frdm_k64f)
  set(OVERLAY_CONFIG overlay-udp.conf)
endif()

# Standard Zephyr application boilerplate.
find_package(Zephyr HINTS $ENV{ZEPHYR_BASE})
project(smp_svr)
```

tejlmand, 2020-04-22:
> This is already supported, but apparently it requires the OVERLAY_CONFIG variable to be set before the zephyr boilerplate is included, like so:
>
> ```
> if (${BOARD} STREQUAL frdm_k64f) [...]

Problem with this approach is that it will not work for deprecated board names where Zephyr will automatically rename the board for the user.
Also it will not work for users who are using aliased board names.

So this is a valid enhancment, all though we have a chicken and hen situation when it comes to coding, cause the boiler plate, and hence overlay config files are parsed as part of `find_package(Zephyr HINTS $ENV{ZEPHYR_BASE})`.
which also handles board aliasing, so this information is needed before `find_package`, but at that time we know nothing about aliased boards.
Note: This was also the situation with old style boilerplate inclusion.

nvlsianpu, 2020-04-22:
@tejlmand ^^ I was aware for that this will be like that (by experience). Thanks for clarification.

fouge, 2022-11-24:
@tejlmand should we close this issue?

nordicjm, 2024-11-19:
Variables are configures so long as they are set before finding the zephyr package, sysbuild is also supported

closed 2024-11-19 by nordicjm