zephyrproject-rtos/zephyr #24554: hal_infineon: Add new module for Infineon XMC HAL layer

parthitce, 2020-04-21:
## Introduction

Support for infineon XMC series ARM cortex M controllers needs/depends on XMCLib. New HAL module needs to be created to address the dependency.

### Problem description

I am submitting the patch to add support for infineon xmc4500 SoC in PR. This PR needs HAL layer for Infineon XMCLib.

### Proposed change

hal_infineon is currently pushed with needed adaptation for Zephyr here: https://github.com/linumiz/hal_infineon

This enables the HAL support for all XMC series SoC's from Infineon

## Detailed RFC

HAL layer can reviewed from here: https://github.com/linumiz/hal_infineon

### Proposed change (Detailed)

NA

### Dependencies

NA

### Concerns and Unresolved Questions

NA

## Alternatives

NA

parthitce, 2020-04-21:
@carlescufi @nashif

carlescufi, 2020-04-21:
The license looks good to me, what is the source of the library @parthitce? Did you download it from an SDK?
@nashif @MaureenHelm @galak any objections? License looks like standard 4-clause BSD to me:
https://github.com/linumiz/hal_infineon/blob/master/XMCLib/drivers/src/xmc4_rtc.c#L12

parthitce, 2020-04-21:
> The license looks good to me, what is the source of the library @parthitce? Did you download it from an SDK?
@carlescufi Yes, it's downloaded from infineon Dave SDK.

Downloaded from: http://dave.infineon.com/Libraries/XMCLib/XMC_Peripheral_Library_v2.1.24.zip

#copied the below statement from [here](https://www.infineon.com/cms/en/product/microcontroller/32-bit-industrial-microcontroller-based-on-arm-cortex-m/32-bit-xmc4000-industrial-microcontroller-arm-cortex-m4/#!tools)
DAVE™ (Version 4) – professional free-of-charge integrated development environment (IDE) supporting the whole development process from evaluation-to-production (E2P).

carlescufi, 2020-04-21:
> @carlescufi Yes, it's downloaded from infineon Dave SDK.

Thanks, could you add any `LICENSE` files the .zip file comes with to your module repo?

parthitce, 2020-04-21:
> > @carlescufi Yes, it's downloaded from infineon Dave SDK.
> 
> Thanks, could you add any `LICENSE` files the .zip file comes with to your module repo?

@carlescufi The Lincense file is included here: https://github.com/linumiz/hal_infineon/blob/master/XMCLib/drivers/License.txt

Should I move to the parent directory of hal_infineon instead?

carlescufi, 2020-04-21:
> Should I move to the parent directory of hal_infineon instead?

If it applies to the whole of the contents, including the `devices/` folder, then yes please.

parthitce, 2020-04-21:
> > Should I move to the parent directory of hal_infineon instead?
> 
> If it applies to the whole of the contents, including the `devices/` folder, then yes please.

Yes, device specific implementation also belongs to the same SDK. I will move the License to parent directory. Thanks.

MaureenHelm, 2020-04-22:
> License looks like standard 4-clause BSD to me

You threw me with this comment. We don't have any 4-clause BSD yet, so that would trigger the license approval process. However, the text actually looks like 3-clause BSD to me.

stephanosio, 2020-04-23:
It is 3-clause BSD license.

The styling in the file headers is unusual and that made it look like "4-clause":
https://github.com/linumiz/hal_infineon/blob/3a27cf18351b2891afbda34ce4a7f528cd12c4d2/XMCLib/drivers/src/xmc1_eru.c#L1-L34

parthitce, 2020-04-23:
I moved the License file to the parent directory here: https://github.com/linumiz/hal_infineon/blob/master/License.txt. Thanks

Yes, this is 3-Clause BSD License (same as [this](https://opensource.org/licenses/BSD-3-Clause)).

carlescufi, 2020-04-26:
@parthitce before we create this repo, can you confirm that you will maintain this platform?

carlescufi, 2020-04-26:
> > License looks like standard 4-clause BSD to me
> 
> You threw me with this comment. We don't have any 4-clause BSD yet, so that would trigger the license approval process. However, the text actually looks like 3-clause BSD to me.

Sorry for the noise!

parthitce, 2020-04-26:
> @parthitce before we create this repo, can you confirm that you will maintain this platform?

@carlescufi Yes, I confirm that I will maintain this platform. Thanks.

parthitce, 2020-05-03:
@carlescufi Ping on this module creation.

carlescufi, 2020-05-06:
@parthitce sorry for the delay, I've been busy with the release. I should be able to create it tomorrow.

carlescufi, 2020-05-06:
@parthitce could you please point us to the hardware documentation for this hardware for the record, so we know we can fix issues by looking at a datasheet or reference manual?

parthitce, 2020-05-06:
> @parthitce could you please point us to the hardware documentation for this hardware for the record, so we know we can fix issues by looking at a datasheet or reference manual?

@carlescufi Current zephyr PR (https://github.com/zephyrproject-rtos/zephyr/pull/24555) is based on xmc4500 SoC alone, but this should be same for all the XMC4xxx series.

TRM XMC4500: https://www.infineon.com/dgdl/Infineon-xmc4500_rm_v1.6_2016-UM-v01_06-EN.pdf?fileId=db3a30433580b3710135a5f8b7bc6d13
Datasheet: https://www.infineon.com/dgdl/Infineon-XMC4500-DS-v01_05-EN.pdf?fileId=5546d46254e133b40154e1b56cbe0123

TRM and Datasheet for all the XMC series is available online.

Overview: https://www.infineon.com/cms/en/product/microcontroller/32-bit-industrial-microcontroller-based-on-arm-cortex-m/#

carlescufi, 2020-05-08:
@parthitce thank you for your patience.
Please push an initial version to:
https://github.com/zephyrproject-rtos/hal_infineon

That said, please make sure that you format the commit similarly to this one:
https://github.com/zephyrproject-rtos/cmsis/commit/0d43487db343e3eacb7175e59103f878599e806a

parthitce, 2020-05-08:
> @parthitce thank you for your patience.
> Please push an initial version to:
> https://github.com/zephyrproject-rtos/hal_infineon
> 
> That said, please make sure that you format the commit similarly to this one:
> [zephyrproject-rtos/cmsis@0d43487](https://github.com/zephyrproject-rtos/cmsis/commit/0d43487db343e3eacb7175e59103f878599e806a)

@carlescufi Thanks, I will re-format the commit message and will push it.

parthitce, 2020-05-09:
https://github.com/zephyrproject-rtos/zephyr/pull/24555 is merged. Closing this issue.

closed 2020-05-09 by parthitce