zephyrproject-rtos/zephyr #24571: #include <new> is not available

nandra, 2020-04-21:
**Describe the bug**
I'm trying to add blinky example using ETL library [link](https://github.com/ETLCPP/etl/blob/master/examples/BlinkList/BlinkList.ino) and facing issue that ETL list.h include ```#include <new>``` and when building using official docker get this:
```
AILED: CMakeFiles/app.dir/src/main.cpp.obj 
ccache /opt/toolchains/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++  -DBUILD_VERSION=zephyr-v2.2.0-1533-g97a5ec0d4f3a -DHSE_VALUE=8000000 -DKERNEL -DSTM32F091xC -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/workdir/samples/etl/src/etl/include/etl -I/workdir/include -Izephyr/include/generated -I/workdir/soc/arm/st_stm32/stm32f0 -I/workdir/drivers -I/workdir/modules/hal/cmsis/CMSIS/Core/Include -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/soc -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include/Legacy -isystem /workdir/lib/libc/minimal/include -isystem /opt/toolchains/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include -isystem /opt/toolchains/zephyr-sdk-0.11.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros/workdir/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m0 -mthumb -mabi=aapcs -march=armv6s-m -imacros/workdir/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wpointer-arith -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/samples/etl=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -nostdinc -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles/app.dir/src/main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c /workdir/samples/etl/src/main.cpp
In file included from /workdir/samples/etl/src/main.cpp:8:
/workdir/samples/etl/src/etl/include/etl/list.h:36:10: fatal error: new: No such file or directory
   36 | #include <new>
      |          ^~~~~
compilation terminated.
[18/120] Building C object zephyr/CMakeFiles/zephyr.dir/soc/arm/st_stm32/stm32f0/soc.c.obj

```
Tried to change to gcc toolchain by exporting variables but issue is the same. Tried also add path directly to include:
-I /opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1

but then it fails with other error:
```
FAILED: CMakeFiles/app.dir/src/main.cpp.obj 
ccache /opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-g++  -DBUILD_VERSION=zephyr-v2.2.0-1533-g97a5ec0d4f3a -DHSE_VALUE=8000000 -DKERNEL -DSTM32F091xC -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/workdir/samples/etl/src/etl/include/etl -I/opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1 -I/workdir/include -Izephyr/include/generated -I/workdir/soc/arm/st_stm32/stm32f0 -I/workdir/drivers -I/workdir/modules/hal/cmsis/CMSIS/Core/Include -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/soc -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include -I/workdir/modules/hal/stm32/stm32cube/stm32f0xx/drivers/include/Legacy -isystem /workdir/lib/libc/minimal/include -isystem /opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem /opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -fcheck-new -std=c++11 -Wno-register -fno-exceptions -fno-rtti -imacros/workdir/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mcpu=cortex-m0 -mthumb -mabi=aapcs -march=armv6s-m -imacros/workdir/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wpointer-arith -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/workdir/samples/etl=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections -nostdinc -MD -MT CMakeFiles/app.dir/src/main.cpp.obj -MF CMakeFiles/app.dir/src/main.cpp.obj.d -o CMakeFiles/app.dir/src/main.cpp.obj -c /workdir/samples/etl/src/main.cpp
In file included from /workdir/samples/etl/src/etl/include/etl/list.h:36,
                 from /workdir/samples/etl/src/main.cpp:8:
/opt/toolchains/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/new:39:10: fatal error: bits/c++config.h: No such file or directory
   39 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

```

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f091rc
3. make
4. See error

**Expected behavior**
Application to be compiled

**Impact**
None

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Linux (docker)
 - Toolchain: Zephyr + GCC
 - Commit: 

**Additional context**
Add any other context about the problem here.

r2r0, 2020-04-22:
I use extensively ETL in my project and no problems so far.
I use Zephyr SDK 0.11.2 on Linux distro (no docker) and more or less present version of Zephyr and STM32WB55 target.
To be able to use ETL I added custom etl_profile.h file and entry in CMakeLists.txt in target_include_directories (path to ETL include directory).

The difference I see in gcc options is in "-I/workdir/samples/etl/src/etl/include/etl" - in my case include path to ETL is without trailing etl.

nandra, 2020-04-22:
@r2r0 I have basically same
CMakeLists.txt:
```
# SPDX-License-Identifier: Apache-2.0

cmake_minimum_required(VERSION 3.13.1)
find_package(Zephyr HINTS $ENV{ZEPHYR_BASE})
project(cpp_synchronization)

target_include_directories(app PUBLIC ${CMAKE_SOURCE_DIR}/src/etl/include/etl)

target_sources(app PRIVATE src/main.cpp)
```

and main.cpp:
```
//***********************************************************************************
// A version of the Blink demo, but with delays stored in two instances of etl::list
//***********************************************************************************

#undef min
#undef max

#include <list.h>
#include <container.h>

/* The devicetree node identifier for the "led0" alias. */
#define LED0_NODE DT_ALIAS(led0)

#if DT_HAS_NODE(LED0_NODE)
#define LED0    DT_GPIO_LABEL(LED0_NODE, gpios)
#define PIN     DT_GPIO_PIN(LED0_NODE, gpios)
#if DT_PHA_HAS_CELL(LED0_NODE, gpios, flags)
#define FLAGS   DT_GPIO_FLAGS(LED0_NODE, gpios)
#endif
#else
/* A build error here means your board isn't set up to blink an LED. */
#error "Unsupported board: led0 devicetree alias is not defined"
#define LED0    ""
#define PIN     0
#endif

struct device* led_pin_setup()
{
	struct device *dev;
        bool led_is_on = true;
	int ret;

	dev = device_get_binding(LED0);
	if (dev == NULL) {
	     return NULL;
	}
	
	ret = gpio_pin_configure(dev, PIN, GPIO_OUTPUT_ACTIVE | FLAGS);
	if (ret < 0) {
		return NULL;
	}
}	

void iterate(const etl::ilist<int>& delays, struct device *dev)
{
    struct device *dev = led_pin_setup()
    etl::ilist<int>::const_iterator itr;

    // Iterate through the list.
    itr = delays.begin();

    while (itr != delays.end())
    {
	    gpio_pin_set(dev, PIN, 1);
	    k_msleep(100);
	    gpio_pin_set(dev, PIN, 0);
	    k_msleep(*itr++);
    }
}

void main()
{
  int delay_times1[] = { 900, 800, 700, 600, 500, 400, 300, 200, 100 };
  int delay_times2[] = { 400, 300, 200, 100 };

  // Fill the first delay list, then reverse it.
  // Notice how we don't have to know the size of the array!
  const size_t size1 = sizeof(etl::array_size(delay_times1));
  etl::list<int, size1> delays1(etl::begin(delay_times1), etl::end(delay_times1));
  delays1.reverse();

  // Fill the second delay list,
  const size_t size2 = sizeof(etl::array_size(delay_times2));
  etl::list<int, size2> delays2(etl::begin(delay_times2), etl::end(delay_times2));

  while (true)
  {
    iterate(delays1, dev);
    iterate(delays2, dev);
  }
}
```
and getting this issue.

r2r0, 2020-04-22:
There are a few (syntax and logical) errors in your code but they do not have nothing in common with ETL nor with Zephyr.
After small changes this code compiles without problems.

BTW> I strongly suggest to not use full path for ETL headers and use <etl/something.h> in your code.
Otherwise you wil get (sooner or later) header names conflicts.

nandra, 2020-04-22:
Yes there could be error as I didn't compile it yet. Basically it's copy'n'paste of this: https://github.com/ETLCPP/etl/blob/master/examples/BlinkList/BlinkList.ino

I'll use etl prefix for etl headers, thanks for suggestions.

carlescufi, 2020-04-22:
@pabigot could you comment on this one?

pabigot, 2020-04-22:
While Zephyr's self-contained minimal libc has some C++ support, `<new>` is part of libcxx so you need to be using newlib.  Including that file works for me with `tests/application_development/libcxx` which enables these options:

```
CONFIG_NEWLIB_LIBC=y
CONFIG_LIB_CPLUSPLUS=y
```

It produces the error you show when the include directive is added to `tests/application_development/cpp` which lacks those options.

nandra, 2020-04-22:
@pabigot thanks this did the trick. :+1:

closed 2020-04-22 by pabigot