zephyrproject-rtos/zephyr #24585: How to read an exsiting file with native posix board?

overheat, 2020-04-22:
Hi, 
I have tried shell access to a LittleFS file system partition in flash on native posix board.
https://docs.zephyrproject.org/latest/samples/subsys/shell/fs/README.html

It works, but how to mount a existing folder on the linux host to zephyr native posix board?
The `lfs` folder under `build/flash` only show up after I mount it in zephyr shell. Then, I try to copy a file to `build/flash/lfs` but failed.

Thanks.
Aaron

vanwinkeljan, 2020-04-22:
> It works, but how to mount a existing folder on the linux host to zephyr native posix board?

You can not mount existing folders to  the native posix board only files can be transferred.
 
> The lfs folder under build/flash only show up after I mount it in zephyr shell.

This is expected behaviour 

> Then, I try to copy a file to build/flash/lfs but failed.

Any error message from zephyr shell?
Note that any LFS restrictions applies.

overheat, 2020-04-22:
@vanwinkeljan 
error message on linux host:
```
$ cp README.rst build/flash/lfs/
cp: error writing 'build/flash/lfs/README.rst': No space left on device
```
error message on native posix board:
```
[00:01:25.960,000] <dbg> littlefs.littlefs_open: alloc 64 file cache: 0
[00:01:25.960,000] <dbg> littlefs.littlefs_open: alloc 64 file cache: 0
[00:01:25.970,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:494: No more free space 6
[00:01:25.970,000] <err> fs: file write error (-28)
```

vanwinkeljan, 2020-04-22:
For me this is working as expected on `native_posix` with the latest master: b29c15ed094b

```
uart:~$ fs mount littlefs /lfs
[00:02:07.300,000] <inf> littlefs: LittleFS version 2.2, disk version 2.0
[00:02:07.300,000] <inf> littlefs: FS at flash_ctrl:0xfc000 is 4 0x1000-byte blocks withe
[00:02:07.300,000] <inf> littlefs: sizes: rd 16 ; pr 16 ; ca 64 ; la 32
[00:02:07.300,000] <err> littlefs: WEST_TOPDIR/modules/fs/littlefs/lfs.c:998: Corrupted }
[00:02:07.300,000] <wrn> littlefs: can't mount (LFS -84); formatting
[00:02:07.300,000] <inf> littlefs: /lfs mounted
```
```
[00:02:30.510,000] <err> fs: failed get file or dir stat (-2)
[00:02:30.510,000] <err> fs: failed get file or dir stat (-2)
uart:~$ fs ls /lfs
README.rst
uart:~$ fs cd /lfs
uart:~$ fs read README.rst
File size: 3899
00000000  2e 2e 20 72 61 77 3a 3a 20 68 74 6d 6c 0a 0a 20 %*c.. raw:: html..
00000010  20 20 3c 61 20 68 72 65 66 3d 22 68 74 74 70 73 %*c  <a href="https
00000020  3a 2f 2f 77 77 77 2e 7a 65 70 68 79 72 70 72 6f %*c://www.zephyrpro
00000030  6a 65 63 74 2e 6f 72 67 22 3e 0a 20 20 20 20 20 %*cject.org">.
00000040  3c 70 20 61 6c 69 67 6e 3d 22 63 65 6e 74 65 72 %*c<p align="center
00000050  22 3e 0a 20 20 20 20 20 20 20 3c 69 6d 67 20 73 %*c">.       <img s
00000060  72 63 3d 22 64 6f 63 2f 69 6d 61 67 65 73 2f 5a %*crc="doc/images/Z
00000070  65 70 68 79 72 2d 50 72 6f 6a 65 63 74 2e 70 6e %*cephyr-Project.pn
00000080  67 22 3e 0a 20 20 20 20 20 3c 2f 70 3e 0a 20 20 %*cg">.     </p>.
00000090  20 3c 2f 61 3e 0a 0a 20 20 20 3c 61 20 68 72 65 %*c </a>..   <a hre
000000a0  66 3d 22 68 74 74 70 73 3a 2f 2f 62 65 73 74 70 %*cf="https://bestp
...
```

overheat, 2020-04-23:
@vanwinkeljan  Thanks.
But I sitll have this `No more free space 6` issue after update code to latest version.
This is my prj.conf
```
# shell/fs
CONFIG_FLASH=y

CONFIG_DEBUG=y
CONFIG_LOG=y
CONFIG_FS_LOG_LEVEL_DBG=y

CONFIG_HEAP_MEM_POOL_SIZE=16384

CONFIG_SHELL=y
CONFIG_SHELL_LOG_LEVEL_INF=y

CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_SHELL=y

CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y

CONFIG_FUSE_FS_ACCESS=y
```

Could do you help to check? or share with me your prj.conf?
I think maybe realted to some `CONFIG_FS_LITTLEFS_XXX` arguments, like `CONFIG_FS_LITTLEFS_FC_MEM_POOL_NUM_BLOCKS` so some else.

vanwinkeljan, 2020-04-23:
I did just use sample `samples/subsys/shell/fs` with one additional config (`CONFIG_LEGACY_TIMEOUT_API`) to get the shell working 

Soo the project config is as follows:
https://github.com/zephyrproject-rtos/zephyr/blob/bd639fc332675278132c57f18a196e23f0287115/samples/subsys/shell/fs/prj.conf#L1-L17
https://github.com/zephyrproject-rtos/zephyr/blob/bd639fc332675278132c57f18a196e23f0287115/samples/subsys/shell/fs/boards/native_posix.conf#L1

overheat, 2020-04-23:
Thanks. I have tried `samples/subsys/shell/fs`, it works for donzes bytes file, like the `VERSION` file under zephyr folder. not works for few KB size file, like the `CODEOWNERS` 
The error is the same.
Maybe relate to my runtime enviroment? I installed Ubuntu 20.04 on my PC.

overheat, 2020-04-23:
@vanwinkeljan `./build/zephyr/zephyr.exe` or `west build -t run` , which one do you use?

vanwinkeljan, 2020-04-23:
> Maybe relate to my runtime enviroment? I installed Ubuntu 20.04 on my PC.

Could be, I did the test run on Ubuntu 18.04.4

> @vanwinkeljan ./build/zephyr/zephyr.exe or west build -t run , which one do you use?

First, I actual change my run directory to `build` and then use `./zephyr/zephyr.exe`

>  not works for few KB size file, like the CODEOWNERS

Actually this is normal as the FS partition is only 16kB and `CODEOWNERS` is 29kB.

https://github.com/zephyrproject-rtos/zephyr/blob/bd639fc332675278132c57f18a196e23f0287115/boards/posix/native_posix/native_posix.dts#L64-L67

overheat, 2020-04-23:
@vanwinkeljan Perfect!
It works for CODEOWNERS (is 29kB) after I change `native_posix.dts` flash storage_partition bigger.
Thanks!

I will close this and change the title, hope help others.

closed 2020-04-23 by overheat

pabigot, 2020-04-25:
See also #24696