zephyrproject-rtos/zephyr #24625: lib: drivers: clock: rtc:  rtc api to maintain calendar time through reboot

sslupsky, 2020-04-23:
I am looking for some feedback on the idea of proposing a new real time clock api.  My use case at the moment is primarily to maintain the calendar time through a reboot event.  At the moment, when a reboot occurs (pin reset, watchdog timeout, brownout, etc), the underlying system timer api mandates that the system tick counter is reset to zero.  The posix clock api and the new date services for the shell allow the user to set the calendar time and this is very useful.  The underlying functions are provided in lib/posix/clock.c.

For the use case I described however, when a reboot event occurs the calendar date is reset to the EPOCH which is problematic for an unattended device that may not have network connectivity or sporadic connectivity.

Some rtc timers have hardware protection of the rtc counter so that the counter is not reset except in the event of a power on reset or a software reset which requires disabling the rtc first.  Thus, a reset as a result of something other than a power event allows the counter to continue counting through the reset event.  I propose that we introduce an api that captures the counter when there is something other than a power on reset, and allows that value to be used as the base time for the rt_clock_base defined in clock.c

Feedback is appreciated.

erwango, 2020-04-23:
@sslupsky You might be interested by https://github.com/zephyrproject-rtos/zephyr/pull/23526

pabigot, 2020-04-27:
This would be possible with #17631 as well, but we need more examples before being able to define an API that will work cross-platform.  (For DS3231 the application must use driver-specific API to determine whether an alarm condition is present that would have triggered the wakeup.)

closed 2020-09-08 by sslupsky

cederom, 2021-05-15:
Hey folks anything good happened in this area? I cannot find much info on RTC, how to set current time, how to obtain timestamp (I need to timestamp samples acquired by sensors), and how to preserve between resets.. any hints appreciacted :-)

sslupsky, 2021-05-17:
What MCU are you using?

cederom, 2021-05-17:
Hello @sslupsky :-) I am using nRF52 chip and will use NRF SDK to control RTC by hand.. I am a little surprised the OS aspiring to be used in critical applications does not have out of the box time-date solution (yet) :-P

sslupsky, 2021-05-20:
@cederom i think most of what you want to do is already there.  I believe the nRF52 has an rtc driver and there are some posix date and time functions to set the date.  To time stamp your samples you can use the kernel time functions such as k_uptime_get().  I have slow sampling requirements so I configure a kernel timer and use that to synchronize my samples.

cederom, 2021-05-20:
Yup this is what I already have working :-) But having this as standard function (even non posix) would be really nice :-) I already eat up around 60% o RAM :-) Thanks :-)

I have created a feature request here: https://github.com/zephyrproject-rtos/zephyr/issues/35333