zephyrproject-rtos/zephyr #24626: USB re-connection fails on SAM E70

stephanosio, 2020-04-23:
**Describe the bug**
On the SAM E70 SoC, the first USB device connection works, but all subsequent re-connections fail (host fails to detect the device and "failed to configure endpoint" is logged).

**To Reproduce**
1. Launch `samples/subsys/usb/cdc_acm` or any other USB device sample with the device USB cable connected.
2. Un-plug the device USB cable and then re-plug.

**Expected behavior**
Host successfully detects the device upon re-connection.

**Screenshots or console output**
Running `samples/subsys/usb/cdc_acm` on `sam_e70_xplained`
```
*** Booting Zephyr OS build zephyr-v2.2.0-1811-g799e725bec40  ***
[00:00:00.006,000] <inf> usb_cdc_acm: USB device suspended
[00:00:00.006,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.009,000] <inf> usb_cdc_acm: USB device suspended
[00:00:00.384,000] <inf> usb_cdc_acm: USB device configured
[00:00:11.017,000] <inf> cdc_acm_echo: DTR set
[00:00:12.025,000] <inf> cdc_acm_echo: Baudrate detected: 115200
>>DISCONNECT USB CABLE<<
[00:00:15.206,000] <inf> usb_cdc_acm: USB device suspended
>>RECONNECT USB CABLE<<
[00:00:16.662,000] <err> usb_device: Failed to configure endpoint 0x81
[00:00:16.662,000] <err> usb_device: Failed to configure endpoint 0x83
[00:00:16.662,000] <err> usb_device: Failed to configure endpoint 0x02
[00:00:16.662,000] <inf> usb_cdc_acm: USB device configured
[00:00:16.850,000] <err> usb_device: Failed to configure endpoint 0x81
[00:00:16.850,000] <err> usb_device: Failed to configure endpoint 0x83
[00:00:16.850,000] <err> usb_device: Failed to configure endpoint 0x02
[00:00:16.850,000] <inf> usb_cdc_acm: USB device configured
[00:00:17.039,000] <err> usb_device: Failed to configure endpoint 0x81
[00:00:17.039,000] <err> usb_device: Failed to configure endpoint 0x83
[00:00:17.039,000] <err> usb_device: Failed to configure endpoint 0x02
[00:00:17.039,000] <inf> usb_cdc_acm: USB device configured
[00:00:17.228,000] <inf> usb_cdc_acm: USB device suspended
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: bd639fc332675278132c57f18a196e23f0287115

**Additional context**
* Tested on `sam_e70_xplained` with the Rev. B chip
* The same issue is expected on `sam_v71_xplained` (cc @nandojve)

stephanosio, 2020-04-23:
cc @aurel32 @mnkp @nandojve @jfischer-phytec-iot

jfischer-phytec-iot, 2020-05-11:
There is just nothing implemented to give/set allocated resource free after re-attach/reset (EORST).  The patch below fix the issue but I guess needs more investigation. Lower priority to low.

```diff
diff --git a/drivers/usb/device/usb_dc_sam.c b/drivers/usb/device/usb_dc_sam.c
index 01673e6ad4..ac8873260b 100644
--- a/drivers/usb/device/usb_dc_sam.c
+++ b/drivers/usb/device/usb_dc_sam.c
@@ -256,6 +256,11 @@ static void usb_dc_isr(void)
                        usb_dc_ep_enable_interrupts(0);
                }
 
+               for (int ep_idx = 1; ep_idx < NUM_OF_EP_MAX; ep_idx++) {
+                       usb_dc_ep_disable(ep_idx);
+                       USBHS->USBHS_DEVEPTCFG[ep_idx] &= ~USBHS_DEVEPTCFG_ALLOC;
+               }
+
                /* Callback function */
                dev_data.status_cb(USB_DC_RESET, NULL);
        }
```

nandojve, 2020-05-12:
Tested @jfischer-phytec-iot patch on SAM_V71_XULT.

procedures:
1- apply patch
2- west build -b sam_v71_xult samples/subsys/usb/cdc_acm
3- west flash
4- connect USB cable on HOST
5- disconnect USB cable from HOST
6- connect USB cable on HOST
7- open minicom at /dev/ttyACM1
8- disconnect USB cable from device
8.1 - confirm "Cannot open /dev/ttyACM1!"
9- connect USB cable on device
9.1 - confirm reconnection
10 - repeat from 8
No errors detected.
```
*** Booting Zephyr OS build v2.3.0-rc1-16-ged2d263e0c23  ***
[00:00:00.005,000] <inf> usb_cdc_acm: USB device suspended
[00:00:00.005,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.008,000] <inf> usb_cdc_acm: USB device suspended
[00:00:06.139,000] <inf> usb_cdc_acm: USB device suspended
[00:00:06.289,000] <inf> usb_cdc_acm: USB device suspended
[00:00:06.304,000] <inf> usb_cdc_acm: USB device suspended
[00:00:08.626,000] <inf> usb_cdc_acm: USB device suspended
[00:00:09.064,000] <inf> usb_cdc_acm: USB device configured
[00:00:17.534,000] <inf> usb_cdc_acm: USB device suspended
[00:00:17.757,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.438,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.464,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.470,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.513,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.521,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.533,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.607,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.615,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.622,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.644,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.671,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.689,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.695,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.703,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.711,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.722,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.733,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.739,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.746,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.755,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.770,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.788,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.796,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.805,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.830,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.868,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.916,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.947,000] <inf> usb_cdc_acm: USB device suspended
[00:00:19.951,000] <inf> usb_cdc_acm: USB device suspended
[00:00:20.020,000] <inf> usb_cdc_acm: USB device suspended
[00:00:20.029,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.429,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.485,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.636,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.784,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.822,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.936,000] <inf> usb_cdc_acm: USB device suspended
[00:00:21.969,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.035,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.053,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.071,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.168,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.368,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.381,000] <inf> usb_cdc_acm: USB device suspended
[00:00:22.396,000] <inf> usb_cdc_acm: USB device suspended
00:00:23.263,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.267,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.334,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.352,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.366,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.484,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.549,000] <inf> usb_cdc_acm: USB device suspended
[00:00:23.586,000] <inf> usb_cdc_acm: USB device suspended
[00:00:24.939,000] <inf> usb_cdc_acm: USB device suspended
[00:00:24.947,000] <inf> usb_cdc_acm: USB device suspended
[00:00:25.364,000] <inf> usb_cdc_acm: USB device configured
[00:00:31.977,000] <inf> usb_cdc_acm: USB device suspended
[00:00:35.988,000] <inf> usb_cdc_acm: USB device configured
[00:00:48.454,000] <inf> cdc_acm_echo: DTR set
[00:00:49.465,000] <inf> cdc_acm_echo: Baudrate detected: 115200
[00:00:54.411,000] <inf> usb_cdc_acm: USB device suspended
[00:00:58.198,000] <inf> usb_cdc_acm: USB device suspended
[00:00:58.616,000] <inf> usb_cdc_acm: USB device configured
[00:01:01.784,000] <inf> usb_cdc_acm: USB device suspended
[00:01:04.386,000] <inf> usb_cdc_acm: USB device suspended
[00:01:04.804,000] <inf> usb_cdc_acm: USB device configured
[00:01:13.637,000] <inf> usb_cdc_acm: USB device suspended
[00:01:21.243,000] <inf> usb_cdc_acm: USB device suspended
[00:01:21.660,000] <inf> usb_cdc_acm: USB device configured
[00:05:45.729,000] <inf> usb_cdc_acm: USB device suspended
```

jfischer-phytec-iot, 2020-05-12:
@nandojve Thanks for testing, there are a lot of suspended events, I have not seen this behavior on same70Xplained.

nandojve, 2020-05-13:
> @nandojve Thanks for testing, there are a lot of suspended events, I have not seen this behavior on same70Xplained.

Indeed there are! I shared all messages to aware everyone. Not sure if that is evidence to do more investigation right now. Well, "today is better than yesterday" and I prefer moving forward.

Off Topic: The only errata documented that affects both Rev A/B silicon (that make sense for stack) is:
```
Issue: No DMA for Endpoint 7
The DMA feature is not available for Pipe/Endpoint 7.
Workaround: None.
```

PR #25243 (2020-05-13): drivers: usb_dc_sam: free endpoint memory on End of Reset event 
Free endpoint memory on End of Reset event (EORST).
Fixes: #24626

I also tweaked logging a little to see what happens during configuration, but I can remove commit f689a5a if not desired.
commit 5a3d4425 (2020-05-13): drivers: usb_dc_sam: free endpoint memory on End of Reset event

Free endpoint memory on End of Reset event (EORST).

Fixes: #24626
closed 2020-05-13 by carlescufi