zephyrproject-rtos/zephyr #24627: tests/subsys/usb/device fails on SAM E54

stephanosio, 2020-04-23:
**Describe the bug**
The `tests/subsys/usb/device` test fails for the SAM E54 SoC.

**To Reproduce**
Run `tests/subsys/usb/device` on `atsame54_xpro`.

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.2.0-1881-gbd639fc33267  ***
D: lock usb_enable_lock mutex
D: set cb, ep: 0x81
D: unlock usb_enable_lock mutex
Running test suite test_device
===================================================================
starting test - test_usb_dc_api_invalid
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: parameter must not be NULL
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
PASS - test_usb_dc_api_invalid
===================================================================
starting test - test_usb_dc_api

    Assertion failed at ../src/main.c:107: test_usb_dc_api: (usb_dc_ep_mps(0x80) not equal to 64)
usb_dc_ep_mps(0x80) failed
FAIL - test_usb_dc_api
===================================================================
starting test - test_usb_dc_api_read_write
E: endpoint index/address out of range
E: endpoint index/address out of range
PASS - test_usb_dc_api_read_write
===================================================================
starting test - test_usb_dc_api_invalid
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: parameter must not be NULL
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
E: endpoint index/address out of range
PASS - test_usb_dc_api_invalid
===================================================================
starting test - test_usb_deconfig
PASS - test_usb_deconfig
===================================================================
starting test - test_usb_disable
PASS - test_usb_disable
===================================================================
Test suite test_device failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.2
 - Commit SHA: bd639fc332675278132c57f18a196e23f0287115

**Additional context**
* USB device operation on the SAM E54 SoC is not affected by this issue (it functions normally).

stephanosio, 2020-04-23:
cc @benpicco @nandojve @jfischer-phytec-iot @aurel32

jfischer-phytec-iot, 2020-05-12:
So SAM E54 SoC uses usb_dc_sam0 driver, but I can not reproduce it on the samr21_xpro. @stephanosio can you check if the issue still present, please?

stephanosio, 2020-05-12:
> So SAM E54 SoC uses usb_dc_sam0 driver, but I can not reproduce it on the samr21_xpro. @stephanosio can you check if the issue still present, please?

@jfischer-phytec-iot This issue is still present. If you don't have a SAM E54 board, I will look into it later.

jfischer-phytec-iot, 2020-05-13:
Something fishy with the platform self? #25231 #25229 #25228 #25227 #25226 #24803

stephanosio, 2020-05-13:
> Something fishy with the platform self?

The platform was only recently added without much validation, so I would say so.

carlescufi, 2020-05-13:
@stephanosio I assigned this to you because @jfischer-phytec-iot does not have a SAM E54 board, although he did check that this was not reproducible on samr21_xpro, which uses the same driver.

nzmichaelh, 2020-06-12:
Hi @stephanosio the SAM0 USB driver has a buffer overflow bug.   Try patching in #26020 and #25879 and see if that fixes the problem.

stephanosio, 2020-07-02:
The problem is fixed in the latest master (2a706d9f209648533d952a1d1d439bccf22cabd3) that has both #26020 and #25879.

closed 2020-07-02 by stephanosio