zephyrproject-rtos/zephyr #24635: tests/counter/counter_basic_api fails on mps2_an385

nordic-krch, 2020-04-23:
**Describe the bug**
Test is failing on CI consistently but passes locally on my setup (Linux VM and WSL).

**To Reproduce**
?

**Expected behavior**
Test shall pass.

**Impact**
Test fails and makes CI red which prevents counter PRs integration.

**Screenshots or console output**
```
*** Booting Zephyr OS build zephyr-v2.2.0-1887-g2a8a54a4e910  ***
Running test suite test_counter
===================================================================
starting test - test_set_top_value_with_alarm
Testing TIMER_0

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/counter/counter_basic_api/src/test_counter.c:198: test_set_top_value_with_alarm_instance: (top_cnt == 5U is false)
TIMER_0: Unexpected number of turnarounds (3).
FAIL - test_set_top_value_with_alarm
===================================================================
starting test - test_single_shot_alarm_notop
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_single_shot_alarm_notop
===================================================================
starting test - test_single_shot_alarm_top
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_single_shot_alarm_top
===================================================================
starting test - test_multiple_alarms
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_multiple_alarms
===================================================================
starting test - test_all_channels
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_all_channels
===================================================================
starting test - test_late_alarm
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_late_alarm
===================================================================
starting test - test_late_alarm_error
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_late_alarm_error
===================================================================
starting test - test_short_relative_alarm
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_short_relative_alarm
===================================================================
starting test - test_cancelled_alarm_does_not_expire
Skipped for TIMER_0
Skipped for TIMER_1
Skipped for DTIMER_0
PASS - test_cancelled_alarm_does_not_expire
===================================================================
starting test - test_set_top_value_without_alarm
Testing TIMER_0
Testing TIMER_1
Testing DTIMER_0
PASS - test_set_top_value_without_alarm
===================================================================
Test suite test_counter failed.
===================================================================
PROJECT EXECUTION FAILED
```

stephanosio, 2020-04-23:
Re-triggered CI on #24374 and #15623 one more time. I have seen this test fail up to three times in a row if I am very unlucky ...

nordic-krch, 2020-04-23:
@stephanosio on CI it fails almost always (if not always). On my setup never :/

stephanosio, 2020-04-23:
> @stephanosio on CI it fails almost always (if not always). On my setup never :/

It happens when CPU load is high (i.e. when building and running multiple tests in parallel, as we do in sanitycheck).

stephanosio, 2020-04-23:
cc @nashif

stephanosio, 2020-04-23:
By the way, enabling QEMU `icount` should eventually get rid of this issue (at least, it did for `qemu_cortex_r5`). See #14173 and #22904.

nordic-krch, 2020-04-23:
i see, so CPU is emulating timer interrupt and interrupt gets delayed when CPU load is high?
any idea how to handle that?

stephanosio, 2020-04-23:
> i see, so CPU is emulating timer interrupt and interrupt gets delayed when CPU load is high?
> any idea how to handle that?

@nordic-krch See the comment above: https://github.com/zephyrproject-rtos/zephyr/issues/24635#issuecomment-618340103

carlescufi, 2020-04-24:
@stephanosio, @andyross  and @wentongwu is this a duplicate of #14173?

stephanosio, 2020-04-24:
> @stephanosio, @andyross and @wentongwu is this a duplicate of #14173?

@carlescufi I would say so.

stephanosio, 2020-04-24:
@wentongwu @andrewboie I wonder if we should merge the `mps2_an385` portion of #22904 separately to make this problem go away, given that `icount` does indeed achieve what we want?

carlescufi, 2020-04-28:
Closing as duplicate of  #14173

closed 2020-04-28 by carlescufi