zephyrproject-rtos/zephyr #24641: inconsistent timer behavior on native platforms

pabigot, 2020-04-23:
A sample I'm trying to prototype using `native_posix{,_64}` is failing because of a behavior difference demonstrated in the program below.

The program has a volatile bool which the main loop spins on.  The bool is to be set after a 50 ms delay using a `k_timer`.

On native_posix it doesn't work: the value is not set so the spin loop doesn't terminate.  The following steps before spinning have the documented effect:
* Busy wait for 100 ms, during which the set should have occurred.  This makes it work.
* Sleep for 100 ms, during which the set should have occurred.  This makes it work.
* Sleep for 1 ms, which should expire before the set occurs.  Spins forever.
* Yield.  Spins forever.
* Just spin right away.  Spins forever.

On real hardware (Nordic particle_xenon, NXP frdm_k64f) the application works without any intervening reschedule-based operations.

So something in native posix is not handling timer interrupts properly.

```
#include <zephyr.h>
#include <sys/printk.h>

static inline void display(const char *tag)
{
	printk("%4u %s\n", k_uptime_get_32(), tag);
}

static volatile bool completed;

static void timer_handler(struct k_timer *timer)
{
	display("timeout");
	completed = true;
}
static struct k_timer timer = Z_TIMER_INITIALIZER(timer, timer_handler, 0);

void main(void)
{
	printk("main thread priority %d\n", k_thread_priority_get(k_current_get()));

	display("setting timer");
	k_timer_start(&timer, K_MSEC(50), K_NO_WAIT);

#if 0 /* Works if long k_busy_wait is used */
	display("waiting 100 ms");
	k_busy_wait(100U * USEC_PER_MSEC);
#elif 0 /* Works if long k_msleep() is used */
	display("sleeping 100 ms");
	k_msleep(100);
#elif 0 /* Does not work if k_msleep() is used */
	display("sleeping 1 ms");
	k_msleep(1);
#elif 0 /* Does not work if k_yield is used */
	display("yielding");
	k_yield();
#else  /* Does not work if none of the above are used */
	display("no wait/yield/sleep");
#endif
	display("spinning");
	u32_t spins = 0;
	while (!completed) {
		++spins;
	}
	display("done");
	printk("Completed with %u spins\n", spins);
}
```

aescolar, 2020-04-24:
Hi @pabigot, I think you are just hitting this basic limitation:
https://docs.zephyrproject.org/latest/boards/posix/native_posix/doc/index.html#important-limitations
In short: In the POSIX arch CPU instructions take 0 simulated time. So your `while(!completed){}` spins in zero simulated time (forever if `completed` was not set before).
Yield does not help you, because as "the CPU is infinitely fast" any other task (which does not `k_busywait()`) will terminate in 0 time, and from your thread point of view, execution will continue after `k_yield()` in the same microsecond in which `k_yield` was called.
If you want to emulate a busy loop in native_posix you need to explicitly busy wait.
You can do something like this:
```C
	while (!completed) {
		++spins;
		#if CONFIG_ARCH_POSIX
		    k_busy_wait(any_number_you_like)
		#endif
	}
```
Note that the default system tick timer for native_posix is 10ms. So even if you advance time

aescolar, 2020-04-24:
So in short, this is a known limitation of the posix arch which can be worked around relatively easily, by explicitly k_busy_waiting. So I hope you will be happy to close the issue and apply the workaround.

pabigot, 2020-04-24:
Yes, that's essentially the workaround I used.

closed 2020-04-24 by pabigot