zephyrproject-rtos/zephyr #24645: naming consistency for kernel object initializer macros

pabigot, 2020-04-23:
What is the policy for naming the initializer value macro for objects defined in kernel.h?

I see the following being used:

Object      | KObj | Recommended | Deprecated
----------- | ---- | ----------- | ----------
FIFO        | no   | `Z_*`       | `K_*`
FUTEX_DATA* | yes  | `Z_*`       |
LIFO        | no   | `_K_*`      | `K_*`
MBOX        | no   | `_K_*`      | `K_*`
MEM_SLAB    | yes  | `_K_*`      | `K_*`
MSGQ        | yes  | `_K_*`      | `K_*`
MUTEX       | yes  | `_K_*`      | `K_*`
PIPE        | yes  | `_K_*`      | `K_*`
POLL_EVENT  | no   | `K_*`       |
POLL_SIGNAL | yes  | `K_*`       |
QUEUE       | yes  | `_K_*`      | `K_*`
SEM         | yes  | `Z_*`       | `K_*`
STACK       | yes  | `_K_*`      | `K_*`
TIMER       | yes  | `Z_*`       | `K_*`
WORK        | no   | `Z_*`       | `K_*`

[*] `FUTEX_DATA` is the initialized object, `FUTEX` is the kernel object.

The selection between `Z_*`, `_K_*`, and `K_*` has no pattern I can immediately discern.  This makes it difficult to figure out how to initialize a kernel object at compile-time (must go to the header and see which spelling is used for the type).

andrewboie, 2020-04-23:
K_ is public.
Z_ is not.
_K_ needs to be changed to Z_, whoever did the naming conversion a couple years ago to rename variables with prefixed underscores to Z_ didn't finish the job.

pabigot, 2020-04-23:
@andrewboie Thanks.  I thought it might be that, but was surprised that I can't statically initialize semaphores and timers without using a non-public API, but I could initialize a mutex.

nashif, 2020-04-23:
>  but was surprised that I can't statically initialize semaphores and timers without using a non-public API.

You should be using `K_SEM_DEFINE` for that?

pabigot, 2020-04-23:
> You should be using `K_SEM_DEFINE` for that?

Thanks for the clue; I see what happened.  This all started because I wanted to statically define a `k_delayed_work` instance, but there is no `K_DELAYED_WORK_DEFINE()` or `K_DELAYED_WORK_INITIALIZER`, so I have to do it with:
```
static k_delayed_work work = {
    .work = Z_WORK_INITIALIZER(work_handler),
};
```

Also I'd planned to have a semaphore inside another structure, for which the `*_DEFINE()` macros don't work.

Probably there should be an issue to finish the conversion of `_K_*` to `Z_*`, and backfill any missing `*_DEFINE()` macros, but that's not this one.  Questions answered; thanks.

PR #24684 (2020-04-24): kernel: rename initializers to be internal
Rename internal macros to use Z_ prefix instead of _K..

Those macros were missed when we did the global renaming activities.

Fixes #24645
commit 45a1d8ac (2020-04-24): kernel: rename initializers to be internal

Rename internal macros to use Z_ prefix instead of _K..

Those macros were missed when we did the global renaming activities.

Fixes #24645
closed 2020-04-23 by pabigot