zephyrproject-rtos/zephyr #24659: tests/portability/cmsis_rtos_v2 failed on reel_board.

chen-png, 2020-04-24:
**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b reel_board -p auto tests/portability/cmsis_rtos_v2/
2.  west flash
3.  See error

**Screenshots or console output**
*** Booting Zephyr OS build zephyr-v2.2.0-1885-gb7897d5aab13  ***
Running test suite test_cmsis_v2_apis

starting test - test_kernel_apis
PASS - test_kernel_apis

starting test - test_delay
PASS - test_delay

starting test - test_thread_apis
PASS - test_thread_apis

starting test - test_thread_apis_dynamic
PASS - test_thread_apis_dynamic

starting test - test_thread_prio
PASS - test_thread_prio

starting test - test_thread_prio_dynamic
PASS - test_thread_prio_dynamic

starting test - test_timer
oneshot_callback (Timer 1) = 1
periodic_callback (Timer 2) = 1
periodic_callback (Timer 2) = 2
periodic_callback (Timer 2) = 3
periodic_callback (Timer 2) = 4
periodic_callback (Timer 2) = 5
PASS - test_timer

starting test - test_mutex
PASS - test_mutex

starting test - test_mutex_lock_timeout
**Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/mutex.c:118: tThread_entry_lock_timeout: status == osErrorResource is false**
PASS - test_mutex_lock_timeout

starting test - test_semaphore
PASS - test_semaphore

starting test - test_mempool
PASS - test_mempool

starting test - test_mempool_dynamic
PASS - test_mempool_dynamic

starting test - test_messageq
**Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/semaphore.c:37: thread_sema: (status == osOK is false)
Semaphore could not be acquired**
FAIL - test_messageq

starting test - test_event_flags_no_wait_timeout
**Assertion failed at WEST_TOPDIR/zephyr/tests/portability/cmsis_rtos_v2/src/msgq.c:41: send_msg_thread: (osMessageQueueGetCount(message_id) not equal to 0)
Something's wrong with osMessageQueueGetCount!**
FAIL - test_event_flags_no_wait_timeout

starting test - test_event_flags_signalled
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x0
E: r0/a1:  0x00000001  r1/a2:  0x00003609  r2/a3:  0x200065b4
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0000cf87
E:  xpsr:  0xa1000000
E: Faulting instruction address (r15/pc): 0x0000cf8e
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200018d4 (unknown)
E: Halting system

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain : zephyr-sdk-0.11.2
 - Commit ID: b7897d5aab

chen-png, 2020-04-24:
git bisect and found the first bad commit is b18b1d435

carlescufi, 2020-04-28:
> git bisect and found the first bad commit is [b18b1d4](https://github.com/zephyrproject-rtos/zephyr/commit/b18b1d4350a4e3c22ba4262b353771d2b34ba522)

Since this commit is completely unrelated, this is probably a red herring and the failure is likely sporadic.

chen-png, 2020-05-07:
also failed on mec15xxevb_assy6853 board.

carlescufi, 2020-05-12:
Closing in favor of https://github.com/zephyrproject-rtos/zephyr/issues/25196

closed 2020-05-12 by carlescufi