zephyrproject-rtos/zephyr #24660: tests/benchmarks/sys_kernel failed on reel_board.

chen-png, 2020-04-24:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p auto tests/benchmarks/sys_kernel/
2. west flash
3. See error

**Screenshots or console output**

*** Booting Zephyr OS build zephyr-v2.2.0-1885-gb7897d5aab13  ***

MODULE: kernel API test
KERNEL VERSION: 0x2026300

Each test below is repeated 10 times;
average time for one iteration is displayed.

TEST CASE: Semaphore #1
TEST COVERAGE:
        k_sem_init
        k_sem_take(K_FOREVER)
        k_sem_give
Starting test. Please wait...
**TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate**
END TEST CASE

TEST CASE: Semaphore #2
TEST COVERAGE:
        k_sem_init
        k_sem_take(TICKS_NONE)
        k_yield
        k_sem_give
Starting test. Please wait...
TEST RESULT: FAILED
**DETAILS: timer tick happened. Results are inaccurate**
END TEST CASE

TEST CASE: Semaphore #3
TEST COVERAGE:
        k_sem_init
        k_sem_take(K_FOREVER)
        k_sem_give
        k_sem_give
        k_sem_take(K_FOREVER)
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #1
TEST COVERAGE:
        k_lifo_init
        k_lifo_get(K_FOREVER)
        k_lifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #2
TEST COVERAGE:
        k_lifo_init
        k_lifo_get(K_FOREVER)
        k_lifo_get(TICKS_NONE)
        k_lifo_put
        k_yield
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: LIFO #3
TEST COVERAGE:
        k_lifo_init
        k_lifo_get(K_FOREVER)
        k_lifo_put
        k_lifo_get(K_FOREVER)
        k_lifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: FIFO #1
TEST COVERAGE:
        k_fifo_init
        k_fifo_get(K_FOREVER)
        k_fifo_put
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

TEST CASE: FIFO #2
TEST COVERAGE:
        k_fifo_init
        k_fifo_get(K_FOREVER)
        k_fifo_get(TICKS_NONE)
        k_fifo_put
        k_yield
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE
...
TEST CASE: Stack #3
TEST COVERAGE:
        k_stack_init
        k_stack_pop(K_FOREVER)
        k_stack_push
        k_stack_pop(K_FOREVER)
        k_stack_push
Starting test. Please wait...
TEST RESULT: FAILED
DETAILS: timer tick happened. Results are inaccurate
END TEST CASE

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain : zephyr-sdk-0.11.2
 - Commit ID: b7897d5aa

nashif, 2020-04-29:
fails on nrf52840dk_nrf52811 and reel_board

nashif, 2020-04-29:
and on nrf9160dk_nrf9160

ioannisg, 2020-05-11:
Failing on nRF5340 as well.
Also the latency test is failing. 
This is not low priority then. Elevating to HIGH for now, to raise awareness (might be Medium though).

carlescufi, 2020-05-18:
@ioannisg and @pabigot could you please spend some time this week on this one?

pabigot, 2020-05-18:
Broken by 10d15d185b63f6616bbb7c9975dff15d52464b11 which seems to have caused problems with `CONFIG_TICKLESS_KERNEL=n` and `CONFIG_SYS_CLOCK_TICKS_PER_SEC=1`.   The benchmark infrastructure tests whether a test runs longer than 1 s by using this configuration.  Fix in #25412.

PR #25412 (2020-05-19): drivers: timer: nrf_rtc: fix TICKLESS=n processing
Some kernel tests use `CONFIG_TICKLESS_KERNEL=n` with
`CONFIG_SYS_CLOCK_TICKS_PER_SEC=1` to detect when a test runs longer
than 1 second.  These tests break if a tick is announced every time a
timeout occurs.  Only announce if the measured duration since the last
tick is at least the duration of a tick.

Fixes #24660
closed 2020-05-19 by carlescufi