zephyrproject-rtos/zephyr #24665: z_cstart memory corruption (ARM CortexM)

ecarusi, 2020-04-24:
The initialization of automatic variable dummy_thread is overwriting non-stack memory. 
In my case the config_levels table get corrupted , but it is code (and memory map) dependant. 

This is happening only with optimizations enabled (either size, speed or debug) and after a quick analysis is cased by stack relocation happening in arch_kernel_init. 

With optimization enabled, the expected struct stack space gets allocated (@1ba4) before stack relocation ,and the offset used after the relocation (@1bdc) causing the memset to corrupt

```
z_cstart:
00001ba4:   push    {lr}
00001ba6:   sub     sp, #116        ; 0x74
42        	u32_t msp = (u32_t)(Z_THREAD_STACK_BUFFER(_interrupt_stack)) +
00001ba8:   ldr     r3, [pc, #96]   ; (0x1c0c <z_cstart+104>)
 400        __ASM volatile ("MSR msp, %0" : : "r" (topOfMainStack) : );
00001baa:   msr     MSP, r3
61        	SCB->CCR |= SCB_CCR_STKALIGN_Msk;
00001bae:   ldr     r4, [pc, #96]   ; (0x1c10 <z_cstart+108>)
00001bb0:   ldr     r3, [r4, #20]
00001bb2:   orr.w   r3, r3, #512    ; 0x200
00001bb6:   str     r3, [r4, #20]
00001bb8:   movs    r3, #224        ; 0xe0
00001bba:   strb.w  r3, [r4, #34]   ; 0x22
00001bbe:   movs    r5, #0
00001bc0:   strb    r5, [r4, #31]
00001bc2:   strb    r5, [r4, #24]
00001bc4:   strb    r5, [r4, #25]
00001bc6:   strb    r5, [r4, #26]
00001bc8:   ldr     r3, [r4, #36]   ; 0x24
00001bca:   orr.w   r3, r3, #458752 ; 0x70000
00001bce:   str     r3, [r4, #36]   ; 0x24
00001bd0:   bl      0xdf0 <z_arm_fault_init>
00001bd4:   bl      0xb4c <z_arm_cpu_idle_init>
146       	SCB->CFSR = SCB_CFSR_USGFAULTSR_Msk |
00001bd8:   mov.w   r3, #4294967295
00001bdc:   str     r3, [r4, #40]   ; 0x28
151       	SCB->HFSR = 0xffffffff;
00001bde:   str     r3, [r4, #44]   ; 0x2c
508       	struct k_thread dummy_thread = {
00001be0:   add     r4, sp, #4
00001be2:   movs    r2, #108        ; 0x6c
151       	SCB->HFSR = 0xffffffff;
00001be4:   mov     r1, r5
00001be6:   mov     r0, r4
00001be8:   bl      0x3292 <memset>
```

Same code in O0

```
          z_cstart:
00003064:   push    {r7, lr}
00003066:   sub     sp, #120        ; 0x78
00003068:   add     r7, sp, #0
500       	gcov_static_init();
0000306a:   bl      0x70a8 <gcov_static_init>
42        	u32_t msp = (u32_t)(Z_THREAD_STACK_BUFFER(_interrupt_stack)) +
0000306e:   ldr     r0, [pc, #168]  ; (0x3118 <z_cstart+180>)
00003070:   bl      0x705e <Z_THREAD_STACK_BUFFER>
00003074:   mov     r3, r0
00003076:   add.w   r3, r3, #2048   ; 0x800
0000307a:   str     r3, [r7, #116]  ; 0x74
0000307c:   ldr     r3, [r7, #116]  ; 0x74
0000307e:   str     r3, [r7, #112]  ; 0x70
 400        __ASM volatile ("MSR msp, %0" : : "r" (topOfMainStack) : );
00003080:   ldr     r3, [r7, #112]  ; 0x70
00003082:   msr     MSP, r3
61        	SCB->CCR |= SCB_CCR_STKALIGN_Msk;
00003086:   ldr     r3, [pc, #148]  ; (0x311c <z_cstart+184>)
00003088:   ldr     r3, [r3, #20]
0000308a:   ldr     r2, [pc, #144]  ; (0x311c <z_cstart+184>)
0000308c:   orr.w   r3, r3, #512    ; 0x200
00003090:   str     r3, [r2, #20]
00003092:   movs    r1, #255        ; 0xff
00003094:   mvn.w   r0, #1
00003098:   bl      0x2e54 <__NVIC_SetPriority>
 94       	NVIC_SetPriority(SVCall_IRQn, _EXC_SVC_PRIO);
0000309c:   movs    r1, #0
0000309e:   mvn.w   r0, #4
000030a2:   bl      0x2e54 <__NVIC_SetPriority>
 98       	NVIC_SetPriority(MemoryManagement_IRQn, _EXC_FAULT_PRIO);
000030a6:   movs    r1, #0
000030a8:   mvn.w   r0, #11
000030ac:   bl      0x2e54 <__NVIC_SetPriority>
 99       	NVIC_SetPriority(BusFault_IRQn, _EXC_FAULT_PRIO);
000030b0:   movs    r1, #0
000030b2:   mvn.w   r0, #10
000030b6:   bl      0x2e54 <__NVIC_SetPriority>
100       	NVIC_SetPriority(UsageFault_IRQn, _EXC_FAULT_PRIO);
000030ba:   movs    r1, #0
000030bc:   mvn.w   r0, #9
000030c0:   bl      0x2e54 <__NVIC_SetPriority>
106       	SCB->SHCSR |= SCB_SHCSR_USGFAULTENA_Msk | SCB_SHCSR_MEMFAULTENA_Msk |
000030c4:   ldr     r3, [pc, #84]   ; (0x311c <z_cstart+184>)
000030c6:   ldr     r3, [r3, #36]   ; 0x24
000030c8:   ldr     r2, [pc, #80]   ; (0x311c <z_cstart+184>)
000030ca:   orr.w   r3, r3, #458752 ; 0x70000
000030ce:   str     r3, [r2, #36]   ; 0x24
41        	z_arm_fault_init();
000030d0:   bl      0x1b38 <z_arm_fault_init>
42        	z_arm_cpu_idle_init();
000030d4:   bl      0x15a0 <z_arm_cpu_idle_init>
146       	SCB->CFSR = SCB_CFSR_USGFAULTSR_Msk |
000030d8:   ldr     r3, [pc, #64]   ; (0x311c <z_cstart+184>)
000030da:   mov.w   r2, #4294967295
000030de:   str     r2, [r3, #40]   ; 0x28
151       	SCB->HFSR = 0xffffffff;
000030e0:   ldr     r3, [pc, #56]   ; (0x311c <z_cstart+184>)
000030e2:   mov.w   r2, #4294967295
000030e6:   str     r2, [r3, #44]   ; 0x2c
508       	struct k_thread dummy_thread = {
000030e8:   adds    r3, r7, #4
000030ea:   movs    r2, #108        ; 0x6c
000030ec:   movs    r1, #0
000030ee:   mov     r0, r3
000030f0:   bl      0x5ee6 <memset>
```

This relies on R7 value which is coherent with the relocation, still not preserved because SP is lower after the relocation.
Running the code up to 0x30e8 and checking registers:
R3: 0x20000f34
SP: 0x200010a0

Quick and dirty workaround is to make the variable static.

Toolchain is gnuarm GCC 9 (2019-q4) and 6 (2017-q2)
Zephyr version is 2.2.0

(MCU is a Nordic nRF52832)

giellamoswhard, 2020-04-24:
We have also tried with zephyr-sdk 0.11.1 and zephyr 2.2.99 master (as of yesterday).

stephanosio, 2020-04-27:
Please provide more details about the bug:

1. Commit hash
2. Exactly what optimisation flag(s) did you use?
3. What sample(s)/test(s) did you build?

stephanosio, 2020-04-27:
I built and ran `samples/hello_world` for `nrf52840dk_nrf52840` with Zephyr SDK 0.11.2 using the following configurations, and I was unable to reproduce this issue:
* No optimisation (-O0)
* "Optimize for size" (-Os)
* "Optimize for speed" (-O2)
* Dangerous optimisation (-O3)

stephanosio, 2020-04-27:
**Synopsis:**
1. `z_cstart` stack frame, including `struct k_thread dummy_thread`, gets allocated before `MSP` is updated by `arch_kernel_init` (-> `z_arm_interrupt_stack_setup`):
https://gist.github.com/stephanosio/2fc45164250bf89370ec108fb43bebe2#file-gistfile1-txt-L10-L13
2. `struct k_thread dummy_thread = { }` initialiser `memset`, assuming `sp` still points to the beginning of the `z_cstart` stack frame, uses `sp + (dummy_thread stack frame offset)` as the `memset` destination pointer:
https://gist.github.com/stephanosio/2fc45164250bf89370ec108fb43bebe2#file-gistfile1-txt-L77

Since `sp` (`msp`) here points to the top of the `z_interrupt_stacks[0]`, we end up nuking the 116 bytes following `z_interrupt_stacks[0] + sizeof(z_interrupt_stacks[0]) + 28`, which happen to be not so important in my case.

@ioannisg Could you double check/validate this? If this is indeed the case, this is a critical bug that requires immediate attention ...

stephanosio, 2020-04-27:
**Note:**
* This bug has been around since v1.14, so the fix for it will need to be backported all the way back to v1.14:
https://github.com/zephyrproject-rtos/zephyr/blame/6627249dc1e1dc3d667081717e2f8935279939d7/kernel/init.c#L470-L475
https://github.com/zephyrproject-rtos/zephyr/blob/035ab2bb72f627422a055380b609d9b1cb5eeaa5/arch/arm/include/kernel_arch_func.h#L37-L39
https://github.com/zephyrproject-rtos/zephyr/blob/035ab2bb72f627422a055380b609d9b1cb5eeaa5/arch/arm/include/cortex_m/stack.h#L51

stephanosio, 2020-04-27:
I see that the Cortex-M boot code is switching to use PSP before calling `z_arm_prep_c`, and there does not seem to be anything that switches the active stack to use MSP:
https://github.com/zephyrproject-rtos/zephyr/blob/035ab2bb72f627422a055380b609d9b1cb5eeaa5/arch/arm/core/cortex_m/reset.S#L91-L98

@ecarusi @giellamoswhard Do you have custom code somewhere that switches to MSP before `z_cstart`?

ecarusi, 2020-04-27:
In my case the MSP was the SP causing the effect described because:

1. At the first run (without the debugger connected) the custom code caused an hard fault
2. By connecting the debugger, **probably due a misconfiguration on my side**, the core was not properly reset
3. After the restart of the code MSP was still the SP
4. The ... structure was corrupted (zero filled) cauing the code to call again the reset vector instead of expected code
5. Starting again from point 3

Long story short, this is not a bug in the code, but a very rare set of conditions caused by a bad J-Link configuration.

ioannisg, 2020-04-27:
thanks for the explanation and the detailed investigation  @ecarusi  @stephanosio

stephanosio, 2020-04-28:
Closing, as the problem has been identified to be a debugger configuration issue.

closed 2020-04-28 by stephanosio