zephyrproject-rtos/zephyr #24674: Cannot generate code coverage report for unit tests using sanitycheck

thoh-ot, 2020-04-24:
**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -p unit_testing --coverage
2. See error

**Expected behavior**
Code coverage rapport for the unit tests in Zephyr.

**Impact**
Annoyance; we wan't to introduce new unit tests of the Zephyr Bluetooth controller code and the coverage rapport is a very valuable tool in that matter.

**Environment (please complete the following information):**
 - OS: Linux (CentOS 7)
 - Toolchain: zephyrsdk-0.11.1
 - Commit: 84817f4906bc2747ea756b327d3cc852baf36d9b

pabigot, 2020-04-28:
AFAIK coverage is only supported when you use `--platform qemu_x86_coverage`.

pabigot, 2020-04-28:
```
tirzah[381]$ sanitycheck -p unit_testing --coverage
Renaming output directory to /mnt/nordic/zp/zephyr/sanity-out.4
INFO    - JOBS: 32
INFO    - Building initial testcase list...
INFO    - 11 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   11/  11  100%  skipped:    0, failed:    0
INFO    - 11 of 11 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 3.59 seconds
INFO    - In total 817 test cases were executed on 1 out of total 243 platforms (0.41%)
INFO    - Generating coverage files...
geninfo: WARNING: no .gcda files found in /mnt/nordic/zp/zephyr/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /mnt/nordic/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/nordic/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/nordic/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/nordic/zp/zephyr/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /mnt/nordic/zp/zephyr/sanity-out/coverage.info
```

PR #24780 (2020-04-29): sanitycheck: support coverage with unit tests
Fix setting coverage for unit tests and link against gcov when coverage
is enabled.

Fixes #24674
commit 051602f4 (2020-04-29): sanitycheck: support coverage with unit tests

Fix setting coverage for unit tests and link against gcov when coverage
is enabled.

Fixes #24674
closed 2020-04-29 by nashif