zephyrproject-rtos/zephyr #24686: can: stm32: Replace master-can-reg by master-can

erwango, 2020-04-24:
**Is your enhancement proposal related to a problem? Please describe.**
In https://github.com/zephyrproject-rtos/zephyr/pull/24396 we introduced an optional dt binding property(int) to provide can1 reg adress to can2.

This was done as a workaround in order to avoid enabling can1 when only can2 is needed to be used, since there was no way to access a property to a node which is not enabled

**Describe the solution you'd like**
As soon it is doable, replace master-can-reg prop(int) by master-can prop(phandle).
Code is available in bellow branch, but it cannot be used due to lack of functionality support at the time I'm raising this issue
https://github.com/erwango/zephyr/tree/dt_can_master_phandle

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2020-04-24:
Current issue is blocked waiting to support access to properties of non enabled nodes with DT api.

lowlander, 2020-09-04:
AFAIK CAN2 can not be enabled without enabling CAN1, because the CAN filters are part of CAN1's clock domain, and so CAN1 has to have its clock configured for CAN2 to work properly.

erwango, 2021-06-28:
@alexanderwachter I've raised this long time ago. Do you think is it still required or we can just close it ?

erwango, 2021-11-03:
Closing as no one complained so far.

closed 2021-11-03 by erwango