zephyrproject-rtos/zephyr #24695: Board IP Can Not Be Set Manually

sercantun, 2020-04-25:
**Describe the bug**
I am using Nucleo F429ZI Board. And trying to run net/sockets/echo server example. 
Problem is that card can not set its own IPv4 nor IPv6 address given with CONFIG_NET_CONFIG_MY_IPV6_ADDR and CONFIG_NET_CONFIG_MY_IPV4_ADDR  parameters. 

Interface 0x20012fe0 (Ethernet) [1]
===================================
Link addr : 02:80:E1:DB:22:89
MTU       : 1500
Ethernet capabilities supported:
        10 Mbits
        100 Mbits
IPv6 unicast addresses (max 3):
        fe80::80:e1ff:fedb:2289 autoconf preferred infinite
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ffdb:2289
IPv6 prefixes (max 2):
        <none>
IPv6 hop limit           : 64
IPv6 base reachable time : 30000
IPv6 reachable time      : 42607
IPv6 retransmit timer    : 0
**IPv4 unicast addresses (max 1):
        <none>
IPv4 multicast addresses (max 1):
        <none>**
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : disabled
DHCPv4 attempts   : 0

jukkar, 2020-04-26:
Note that the config option is named `CONFIG_NET_CONFIG_PEER_IPV4_ADDR`, are you missing `CONFIG_`?

sercantun, 2020-04-27:
Yes, the configuration is set with CONFIG_NET_CONFIG_PEER_IPV4_ADDR.

jukkar, 2020-04-27:
How you have modified the conf file, the upstream conf file works just fine with the board you mentioned.

carlescufi, 2020-04-28:
@sercantun please post the whole `build/zephyr/.config` file.

sercantun, 2020-04-30:
.config file can not be generated because of the compile error.

Compile stops here;
Error: Aborting due to non-whitelisted Kconfig warning
../zephyrproject/zephyr/samples/net/sockets/echo_server/prj.conf:47: warning: ignoring
malformed line ' CONFIG_NET_CONFIG_SETTINGS=y''.

jukkar, 2020-04-30:
> Error: Aborting due to non-whitelisted Kconfig warning
> ../zephyrproject/zephyr/samples/net/sockets/echo_server/prj.conf:47: warning: ignoring
> malformed line ' CONFIG_NET_CONFIG_SETTINGS=y''.

Do you have extra space in start of the config entry, the error print indicates that.

sercantun, 2020-05-09:
Thank you all for your support. Problem is solved when I re-install zephyr in another directory. And board had a static IP.

closed 2020-05-10 by jukkar