zephyrproject-rtos/zephyr #24700: tests/drivers/kscan/kscan_api failed on mimxrt1050_evk board.

chen-png, 2020-04-26:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b mimxrt1050_evk -p auto tests/drivers/kscan/kscan_api
2. west flash
3. See error

**Screenshots or console output**
E: Could not enable
*** Booting Zephyr OS build zephyr-v2.2.0-1974-g4d7d3a25c127  ***
Running test suite kscan_basic_test

starting test - test_init_callback
**Cannot get KBSCAN device
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/kscan/kscan_api/src/test_kscan.c:87: test_init_callback: test_kb_callback() == TC_PASS is false**
FAIL - test_init_callback

starting test - test_control_callback
**Cannot get KBSCAN device
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/kscan/kscan_api/src/test_kscan.c:97: test_control_callback: test_disable_enable_callback() == TC_PASS is false**
FAIL - test_control_callback

Test suite kscan_basic_test failed.

PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.2
 - Commit ID: 4d7d3a25c1

MaureenHelm, 2020-06-25:
I can't reproduce this on 4d7d3a2 or a more recent 22be0d8fd5902da11c9bf55a44012abc7c2cf1c1. Is this still an issue?

chen-png, 2020-07-03:
> Is this still an issue?

yes, it still failed with latest commit ab3e778f47

*** Booting Zephyr OS build zephyr-v2.3.0-720-gab3e778f47d8  ***
Running test suite kscan_basic_test

START - test_init_callback
Cannot get KBSCAN device
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/kscan/kscan_api/src/test_kscan.c:87: test_init_callback: test_kb_callback() == TC_PASS is false
FAIL - test_init_callback

START - test_control_callback
Cannot get KBSCAN device
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/kscan/kscan_api/src/test_kscan.c:97: test_control_callback: test_disable_enable_callback() == TC_PASS is false
FAIL - test_control_callback

Test suite kscan_basic_test failed.

PROJECT EXECUTION FAILED

MaureenHelm, 2020-07-06:
I see the problem now. There is an [LCD panel](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/4-3-lcd-panel:RK043FN02H-CT) for this board that I bet you don't have in your board farm. Without it, the kscan driver init will fail and `device_get_binding(KSCAN_DEV_NAME)` will return NULL.

@nashif can you recommend how to filter this in `testcase.yaml`? Add a harness? Not sure I want to blacklist the board since we may have the display attached in other board farms.

cc: @hakehuang

MaureenHelm, 2020-07-28:
@nashif ping

hakehuang, 2020-08-18:
my log shows:
```
    *** Booting Zephyr OS version 2.3.99  ***
    Running test suite kscan_basic_test
    ===================================================================
    START - test_init_callback
     PASS - test_init_callback
    ===================================================================
    START - test_control_callback
     PASS - test_control_callback
    ===================================================================
    Test suite kscan_basic_test succeeded
    ===================================================================
    PROJECT EXECUTION SUCCESSFUL

```

MaureenHelm, 2020-08-19:
@hakehuang do you have the LCD attached?

hakehuang, 2020-08-20:
@MaureenHelm  yes. I have the LCD attached.

carlescufi, 2020-09-08:
Will be fixed in #28191

PR #28191 (2020-09-08): samples: Add fixtures for samples interacting with keyboard and mouse
Two samples require interacting with keyboard and mouse, and that
means necessary to provide correct fixture before run of that samples
in sanitycheck system. To skip or run the sample according to the
provided fixture in map-file necessary to add fixture definitions
into sample.yaml files
Fixes #28146 
Fixes #28149 
Fixes #24700
closed 2020-09-08 by nashif